sy2002/QNICE-FPGA #1: UHS standard

AnoushkaU, 2020-06-05:
Hello!

Could you please suggest which Xilinx FPGA I can use to implement this design for a UHS standard card? 

Thank you!

sy2002, 2020-06-05:
@AnoushkaU I used a Xilinx Artix-7 board but I guess it will also work on a lot of other FPGAs. [Here is a link about our setup](https://github.com/sy2002/QNICE-FPGA/tree/1077288aec32b6e23b59402d4161eb513119c49c/hw#nexys-4-ddr-and-nexys-a7).

closed 2020-06-05 by sy2002

AnoushkaU, 2020-06-05:
> @AnoushkaU I used a Xilinx Artix-7 board but I guess it will also work on a lot of other FPGAs. [Here is a link about our setup](https://github.com/sy2002/QNICE-FPGA/tree/1077288aec32b6e23b59402d4161eb513119c49c/hw#nexys-4-ddr-and-nexys-a7).

Thank you very much for your reply. Your setup looks great! Do you know if it will support UHS standards? For example, a SanDisk or Transcend SDHC card with UHS-I support?

sy2002, 2020-06-05:
Please have a look at this post from me:

https://github.com/ibm2030/SimpleSDHC/issues/6#issuecomment-629204552

What I found out in general is, that SD Cards do not really like the SPI mode. They kind of work, but if you want a truly stable SD Card implementation, then you need to implement native mode. The challenge here is, that the documentation is not freely available and needs to be licensed.

I made the experience that you really need to do trial and error: Some cards work and some don't. I therefore decided to accept the fact, that my implementation will not work with all SD Cards, but only with some. I went on to buy something like 30 different SD Cards on Amazon and tested them one after another. I found out some patterns, which one work and which don't and documented them here:

https://github.com/sy2002/QNICE-FPGA/blob/master/doc/constraints.txt#L13

Bottom line: My code is the result of trial and error and therefore is not not working with all brands and/or all card sizes. You can have a look at the comment section here to find out what I did:

https://github.com/sy2002/QNICE-FPGA/blob/master/vhdl/sd_spi.vhd#L106

In the end I thought: I can live with it: I have a lists of brands and card models that work and this is good enough for our hobby project...

Sorry, that this is all that I can say, but SD Cards via SPI is black magic. I think this is done intentionally, so that the SD Card consortium can sell licenses on the "native mode" documentation. :-) They just do not consider hobbyists like us as a relevant market.

AnoushkaU, 2020-06-05:
> Please have a look at this post from me:
> 
> [ibm2030/SimpleSDHC#6 (comment)](https://github.com/ibm2030/SimpleSDHC/issues/6#issuecomment-629204552)
> 
> What I found out in general is, that SD Cards do not really like the SPI mode. They kind of work, but if you want a truly stable SD Card implementation, then you need to implement native mode. The challenge here is, that the documentation is not freely available and needs to be licensed.
> 
> I made the experience that you really need to do trial and error: Some cards work and some don't. I therefore decided to accept the fact, that my implementation will not work with all SD Cards, but only with some. I went on to buy something like 30 different SD Cards on Amazon and tested them one after another. I found out some patterns, which one work and which don't and documented them here:
> 
> https://github.com/sy2002/QNICE-FPGA/blob/master/doc/constraints.txt#L13
> 
> Bottom line: My code is the result of trial and error and therefore is not not working with all brands and/or all card sizes. You can have a look at the comment section here to find out what I did:
> 
> https://github.com/sy2002/QNICE-FPGA/blob/master/vhdl/sd_spi.vhd#L106
> 
> In the end I thought: I can live with it: I have a lists of brands and card models that work and this is good enough for our hobby project...
> 
> Sorry, that this is all that I can say, but SD Cards via SPI is black magic. I think this is done intentionally, so that the SD Card consortium can sell licenses on the "native mode" documentation. :-) They just do not consider hobbyists like us as a relevant market.

Thank you so much for your detailed explanation! And yes, SPI is a real headache here! I will try to use one of the cards that you have already tested as working. :)