sy2002/QNICE-FPGA #11: Add qtransfer as standard functionality to the Monitor

sy2002, 2020-07-28:
Add qtransfer as standard functionality to the Monitor

sy2002, 2020-08-05:
Added to scope of issue #28, i.e. release 1.6

sy2002, 2020-08-08:
Do that in the `develop` branch after all the other branches were merged back down to `develop`

sy2002, 2020-08-18:
* After this is done, we also need to update the documentation about qtransfer in `doc/README.md`.
* Also, from this moment on, we might build it by default in `tools/make-toolchain.sh` and only if the serial library is missing, we would output the instructoins of how to build it manually (as we do it today).

MJoergen, 2020-08-19:
> Also, from this moment on, we might build it by default in `tools/make-toolchain.sh` and only if the serial library is missing, we would output the instructoins of how to build it manually (as we do it today).

Yes, please do that. It seems strange that it is not built automatically.

sy2002, 2020-09-08:
done

closed 2020-09-08 by sy2002