sy2002/QNICE-FPGA #5: .DW with undefined label yields no error

bernd-ulmann, 2020-07-28:
The assembler issues no error message in case of a .DW directive which uses an undefined label. Something like

.ORG 0x8000
.DW BLA

yields 0x7FFE in address 0x8000 while it should yield an error.

bernd-ulmann, 2020-07-28:
This problem was caused by str2int which silently returned nonsense if called with a non-number string.

Therefore str2int has now been extended with a second parameter *error which can be used to flag an error condition to the calling program. 

This error state should be examined at every location where str2int is called. As of now, it is only done in the second section of the .DW handling code.

sy2002, 2020-07-28:
Solved by commit dea4acac5b7b344fa30575e59749fe15440b8a6a

closed 2020-07-28 by sy2002