sy2002/QNICE-FPGA #20: Make the native assembler

sy2002, 2020-07-31:


MJoergen, 2020-08-08:
When we write a new assembler it would be nice if it supported macro instructions, i.e. defining a macro consisting of small snippets of assembly. Something similar to the 6502 macro assembler, see sections 6 and 12 here: [https://cc65.github.io/doc/ca65.html](https://cc65.github.io/doc/ca65.html)

sy2002, 2020-08-08:
@MJoergen Good point: Macros for the native toolchain's assembler would be nice, so @bernd-ulmann if you added a preprocessor, that would be a nice additional point. But @MJoergen Did you know: **We already do have a QNICE assembler, that is capable of macros**: [VASM](http://sun.hasenbraten.de/vasm/): When you go to the website, you see QNICE mentioned there. You find infos about how to use it in QNICE-FGA [here](http://sun.hasenbraten.de/vasm/), scroll down to the section *VASM assembler*.

MJoergen, 2020-08-08:
Oh, I didn't know about VASM. Thank you for the hint.

sy2002, 2020-08-08:
Sorry, wrong link in above comment (now edited). You find how to use it in QNICE-FPGA in our README.md, here:

https://github.com/sy2002/QNICE-FPGA/tree/dev-int#programming-in-assembler

bernd-ulmann, 2020-08-08:
Simple macros are also possible with the existing qasm using the C-preprocessor which is called prior the assembler run by asm. Maybe better than nothing at the moment. :-)