sy2002/QNICE-FPGA #24: Fix emulator bug in software interrupt (INT)

sy2002, 2020-08-01:
@bernd-ulmann The emulator seems to struggle either with post increments at software interrupts or with two software interrupts in a row (or with something else)

Switch to branch `dev-int` and pull: I enhanced `test_programs/int_test.asm` like this:

```
        [...]
        MOVE    PREDECIND_1, R8 ; Test indirect ISR address with predecrement
        INT     @--R8

        INT     @R8++           ; test postincrement during INT
        INT     @R8

        MOVE    ITEST_3, R8
        SYSCALL(puts, 1)
        SYSCALL(exit, 1)
```

And while doing so, I also added an appropriate ISR to which `@R8++` should point to and made sure that R8 is pushed on the stack during the PRE ISR, so that the correct R8 should still be available when doing the post increment.

The resulting output of the new test program should be:

```
QMON> CONTROL/RUN ADDRESS=8000
Start
ISR_ABS
ISR_REG
ISR_IND
ISR_PRE
ISR_PRE
ISR_POST
End

QMON> 
```

But in fact, the program outputs three `ISR_PRE` in a row.

bernd-ulmann, 2020-08-01:
Fixed - I forgot to update the destination register in case of an INT instruction.

closed 2020-08-01 by bernd-ulmann

sy2002, 2020-08-01:
Re-tested successfully: Good job! :-)