sy2002/QNICE-FPGA #25: Fix emulator bug that prevents the timer from starting

sy2002, 2020-08-01:
In branch **dev-int**: There seems to be an emulator bug that prevents the timer from starting after a write/read register test:

I added a register write/read test to `test_programs/timer_test.asm`: When running it at `0xE000` at first all registers are filled with some dummy values and then the dummy values are tested.

The write/read test itself passes, but after the successful write/read test, the timer interrupt's ISR is never called.

bernd-ulmann, 2020-08-02:
It is even stranger than that. The timer interrupt can be activated once but not a second time:

QMON> MEMORY/LOAD - ENTER ADDRESS/VALUE PAIRS, TERMINATE WITH CTRL-E
0E00000F800E0010FF300E00200F840E0030FF310E00400F880E0050FF320E00600F8C0E0070FF330E00800F900E0090FF340E00A00F940E00B0FF350E00C00F810E00D0FFFF0E00E00F850E00F0EEEE0E01000F890E0110DDDD0E01200F8D0E0130CCCC0E01400F910E0150BBBB0E01600F950E0170AAAA0E0180CF890E0190DDDD0E01A0FFAB0E01B000610E01C00F890E01D000000E01E0CF950E01F0AAAA0E0200FFAB0E0210005B0E02200F950E023000000E0240CF810E0250FFFF0E0260FFAB0E027000550E0280CF850E0290EEEE0E02A0FFAB0E02B000510E02C0CF8D0E02D0CCCC0E02E0FFAB0E02F0004D0E0300CF910E0310BBBB0E0320FFAB0E033000490E03400FA00E0350E08A0E0360FF900E037000080E03800F800E0390FF300E03A00F820E03B000640E03C00F820E03D003E80E03E00F820E03F0E0830E0400FF900E041000160E042000540E043000690E0440006D0E045000650E046000720E047000200E048000490E0490006E0E04A000740E04B000650E04C000720E04D000720E04E000750E04F000700E050000740E051000200E052000440E053000650E054000760E055000690E056000630E057000650E0580003A0E059000200E05A000520E05B000650E05C000670E05D000690E05E000730E05F000740E060000650E061000720E062000200E063000770E064000720E065000690E066000740E067000650E0680002F0E069000720E06A000650E06B000610E06C000640E06D000200E06E000740E06F000650E070000730E071000740E072000200E073000660E074000610E075000690E0760006C0E077000650E078000640E0790002E0E07A0000D0E07B0000A0E07C000000E07D00FA00E07E0E0420E07F0FF900E080000080E0810FF900E082000160E083008370E08400FA00E0850E0C00E0860FF900E087000080E08800DA00E0890E0400E08A000530E08B000650E08C000740E08D000750E08E000700E08F000200E090000740E091000690E0920006D0E093000650E094000720E095000200E096000300E097000200E098000740E0990006F0E09A000200E09B000690E09C0006E0E09D000740E09E000650E09F000720E0A0000720E0A1000750E0A2000700E0A3000740E0A4000200E0A5000650E0A6000760E0A7000650E0A8000720E0A9000790E0AA000200E0AB000310E0AC000300E0AD000300E0AE000300E0AF000200E0B00006D0E0B1000690E0B20006C0E0B30006C0E0B4000690E0B5000730E0B6000650E0B7000630E0B80006F0E0B90006E0E0BA000640E0BB000730E0BC0002E0E0BD0000D0E0BE0000A0E0BF000000E0C0000540E0C1000690E0C20006D0E0C3000650E0C4000720E0C5000200E0C6000300E0C7000200E0C8000680E0C9000610E0CA000730E0CB000200E0CC000690E0CD000730E0CE000730E0CF000750E0D0000650E0D1000640E0D2000200E0D3000610E0D40006E0E0D5000200E0D6000690E0D70006E0E0D8000740E0D9000650E0DA000720E0DB000720E0DC000750E0DD000700E0DE000740E0DF000200E0E0000720E0E1000650E0E2000710E0E3000750E0E4000650E0E5000730E0E6000740E0E7000210E0E80000D0E0E90000A0E0EA000000E10000F800E1010FF320E10200F810E103000000E1040FF900E10500016
QMON> MEMORY/DISASSEMBLE START ADDRESS=E000 END ADDRESS=E010
E000 0F80 MOVE    FF30 R0
E002 0F84 MOVE    FF31 R1
E004 0F88 MOVE    FF32 R2
E006 0F8C MOVE    FF33 R3
E008 0F90 MOVE    FF34 R4
E00A 0F94 MOVE    FF35 R5
E00C 0F81 MOVE    FFFF @R0
E00E 0F85 MOVE    EEEE @R1
E010 0F89 MOVE    DDDD @R2
QMON> CONTROL/RUN ADDRESS=E000
Setup timer 0 to interrupt every 1000 milliseconds.

QMON> Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
Timer 0 has issued an interrupt request!
CONTROL/RUN ADDRESS=Timer 0 has issued an interrupt request!
E1Timer 0 has issued an interrupt request!
00

QMON> CONTROL/RUN ADDRESS=E000
Setup timer 0 to interrupt every 1000 milliseconds.

QMON> CONTROL/RUN ADDRESS=E000
Setup timer 0 to interrupt every 1000 milliseconds.

QMON>
	Aborted by CTRL-C!

bernd-ulmann, 2020-08-02:
It works now but I fear that there might lurk a nasty problem in the dark of the (way too simple as it seems) timer implementation. usleep and signals do not mix too well. If we encounter problems here we might need to switch to nanosleep and reschedule such sleeps if the planned interval has not completely elapsed... I will close this issue for now - if further problems manifest themselves we can reopen it. :-)

closed 2020-08-02 by bernd-ulmann