sy2002/QNICE-FPGA #27: Review/update/correct CONTRIBUTING.md

sy2002, 2020-08-03:
I created the [CONTRIBUTING.md](https://github.com/sy2002/QNICE-FPGA/blob/master/CONTRIBUTING.md) at the `master` branch, because I thought it belongs there and it might take quite a while until we make our next release (aka update `master` with real code).

@MJoergen and @bernd-ulmann Please review [CONTRIBUTING.md](https://github.com/sy2002/QNICE-FPGA/blob/master/CONTRIBUTING.md): If you find obvious errors, please correct them directly in the file. 

Also, if you have useful additions, please update the file directly.

If there are controversial points - let's discuss them here in this issue.

@MJoergen I invited you to this repo for full write acccess - welcome! 😃 

P.S. Some of the links in [CONTRIBUTING.md](https://github.com/sy2002/QNICE-FPGA/blob/master/CONTRIBUTING.md) are right now (as a temporary compromise solution) pointing to files in the `develop` branch, just because - as written above - it might take a while until we make the next release `V1.6`.

MJoergen, 2020-08-04:
Great start! Thank you for the write access :+1: 

Here are some thoughts:

1.
> We do not have a lot of rules, neither do we have a detailed written code of conduct or even written coding guidelines. As written above: This is meant to be a relaxed project :-) Nevertheless, here are some basics that you might want to follow when contributing.

This seems too vague, particularly the "you might want to follow". Let's rephrase that as "we expect you to follow" or something similar.

2.
The list of documentation should be moved to the top-level README.md

sy2002, 2020-08-04:
Sounds good, Michael. Please do that.

MJoergen, 2020-08-04:
Done :-)

closed 2020-08-05 by sy2002