sy2002/QNICE-FPGA #28: Release 1.6: Scope

sy2002, 2020-08-05:
The latest stable release is V1.5. Currently, we have a ton of loose ends, so let us use this issue to define the **scope of release 1.6** and track the progress towards release 1.6. Let's not start even more loose ends (such as removing the tristates everywhere) right now: We might soon open a second issue "Release 1.7: Scope" where we put all the stuff that takes longer.

## Original intend of release 1.6 (branch `develop`)

Originally, I started release 1.6 with the goal to port QNICE-FPGA to the [MEGA65](https://mega65.org/). Before being able to do that, I needed to port it to Vivado first and before being able to do that, I needed to create the `hw` folder including `hw/README.md` where I did split the multiple FPGA hardware types and the build environment. Also the `vhdl/hw` folder was created to host hardware specific code. When porting QNICE-FPGA to the MEGA65 I also improved a lot of documentation. And I made the whole toolchain Linux friendly.

Have a look at the work-in-progress [VERSIONS.txt](https://github.com/sy2002/QNICE-FPGA/blob/develop/VERSIONS.txt) in the develop branch, it contains a rough "What has already been done" and what is still to be done list.

### What is already finshed from the original intend of V1.6

Here is the summary of what has already been done for V1.6 which is part of the `develop` branch:

* MEGA65: Ported QNICE-FPGA to the MEGA65 (https://mega65.org/)
* UART: Added a 32 Byte FIFO to the UART. This was necessary because the
  MEGA65 serial over JTag does not support RTS/CTS and therefore we needed
  more stability. Now RTS/CTS should be unnecessary in most cases.
* INSTRUCTION COUNTER: Added. Counts all instructions executed by the CPU.
  Can be used for performance testing as shown in
  test_programs/q-tris_perf_test.asm and test_programs/mandel_perf_test.asm.
* EMULATOR: Made the WebAssembly target compatible with Emscripten 1.39.14
  (and newer and it still works with 1.39.10 and some of the older versions,
  but not with the versions in between). The background of this phenomenon is
  explained here: https://github.com/emscripten-core/emscripten/issues/10746
* TOOLCHAIN: qtransfer: Convenient mechanism for transferring software
  (.out files) from the host computer to QNICE-FPGA. Details: doc/README.md
* TOOLCHAIN: Added Xilinx Vivado as an alternative IDE to work with.
  Structured all hardware dependencies in the new folder 'hw' and added
  a documentation in 'hw/README.md'.
* TOOLCHAIN: Added Ubuntu Linux support, which means it should also run on
  any other Debian based distribution. You might want to install xclip for
  more convenience (.out copied to clipboard after assembler ran).
* DOCUMENTATION: Added doc/README.md, hw/README.md & vhdl/hw/MEGA65/README.md

### What needs still to be done to finalize the original intend of V1.6?

* EMULATOR: Correct emulator speed: Using the cycle and instruction counter, we found out that the actual MIPS of the QNICE hardware is not equal to 12.96 MIPS. Correct that in the VGA and WASM versions of the emulator.
* MEGA65 HyperRAM: The HyperRAM driver is not fully stable yet: Stabilize it (8bit native HyperRAM version) and make the 16bit QNICE version stable. Update the appropriate test programs. If time and energy left: Finalize "The Matrix" ASCII video as a demo for the HyperRAM
* Make sure that the PORE texts are correct (for the MEGA65 version as well as for the standard version)
* Restructure dist_kit and have .bit and .cor files ready and then update all the links for example in the hw/README.md and doc/README.md (are there more?)
* Make sure to update the Getting Started section to reflect all the news about ISE and Vivado and also about the platforms. Have also a MEGA65.bit and a MEGA65.cor in the dist_kit and mention it in the Getting Started  section

## Interrupt system for V1.6 (branch `dev-int`)

In branch `dev-int` we are developing the QNICE interrupt system as it is described on pages 9, 12, 23-25 in [this updated version of the intro document](https://github.com/sy2002/QNICE-FPGA/blob/dev-int/doc/intro/qnice_intro.pdf).

### Interrupt system: Already finished

* Assembler is updated to support the new instructions
* Intro document, programming card updated
* Emulator improved to support the interrupt system
* Two test programs are written: `test_programs/int_test.asm` to test the software interrupts in all forms and shapes and `test_programs/timer_test.asm` to test hardware interrupts using the timer
* Hardware: Development testbed created in `vhdl/sim`: Everything related with the development of the interrupt system is called `dev_int*`: The `asm` file is the test progrram, the `dev_int.vhd` file is the top file for the simulation.
* Hardware: The CPU successfully supports hardware and software interrupts

### Interrupt system: TODOs for V1.6

* Hardware: Finalize the work-in-progress (WIP) hardware timer module `timer_module.vhd` and `timer.vhd`, particularly: Implement the correct Daisy Chaining mechanism and behaviour and rename the `left` `right` logic of the Daisy Chain in `in` and `out`
* Test the hardware (at least using the above-mentioned test programs)
* Is there a cool demo program that we could write to show off the interrupt system?

## Change the memory map and refactor MMIO registers (branch `dev-io-addr`)

The goal is to have a more structured (and hardware implementation friendly) memory map. Issue #18 is tracking the progress.

### What has been done already

* Define new memory map in `monitor/sysdef.asm`
* Fix legacy test programs that used hardcoded MMIO addresses
* Fix the emulator by including ../dist_kit/sysdef.h instead of using hardcoded own #defines to access registers
* Add MEGA65 MMIO to sysdef.asm
* Updated programming card
* Updated README.md

### What remains to be done

* Refactor `keyboard.vhd`
* Update MMIO hardware
* Search for old addresses in the comments in the VHDL files (e.g. keyboard.vhd) and replace them all
* Test the updated MMIO hardware
* MEGA65: Update the hardware, update the test programs, retest on hardware
* Recompile standard C library
* Perform some smoke tests for the recompiled standard C library
* Recompile all /qbin demos and tools and check-in the new /qbin
* Create a new disk image as described [here](https://github.com/sy2002/QNICE-FPGA/issues/18#issuecomment-667398346)
* Modify `emulator/run_vga.bash` to support the new disk image
* Put the new disk image also to the two standard SD card (Nexys4DDR and MEGA65) and test them there
* Do not update the website http://qnice-fpga.com/emulator.html as nobody would notice the MMIO change there, BUT: We need to update the source code to download the V1.6 version of the disk image. Then: Do a smoke test locally by recompiling and testing the WebAssembly version of the emulator.

MJoergen, 2020-08-05:
This is great, thanks!

We should open "Release 1.7: Scope" as well to discuss things that are likely to be part of that release. This could include e.g.
* Add multicolour support in VGA (Issue #17).
* BS/DEL when entering hex digits (and at other reasonable places) (Issue #13).
* Other issues ?

Other stuff to consider is:
* Replace tristate registers with separate input/output (do we still need this? I guess we should check if that solves Issue #4)

Oh, and the source files in the vhdl directory have inconsistent line endings: Some are pure DOS (i.e. CR+LF), some are pure Linux (i.e. LF only) and some files have mixed line endings. This really bothers me, and I would ideally like all source files to use the same convention for line endings. I don't have strong feelings for either DOS or Linux mode, but I do value consistency.

MJoergen, 2020-08-05:
Alternatively (or simultaneously) we could write in each open issue an expected target release (no promises, only for planning purposes). But this will give a sense of priority and an impression of a road map.

MJoergen, 2020-08-05:
### Done

- Refactor keyboard.vhd
- Update MMIO hardware
- Search for old addresses in the comments in the VHDL files (e.g. keyboard.vhd) and replace them all
- Test the updated MMIO hardware

There are a number of files in hw/MEGA65 that I haven't touched. Are they needed for the MEGA65 platform? Could you (@sy2002 ) update and test these files ?

sy2002, 2020-08-05:
@MJoergen Thank you for the keyboard refactor and tests. Sounds great. Here are some answers to your points:

I like the idea of tags, you can filter them directly in the issues list. So I added tags to nearly all of the issues. @bernd-ulmann needs to decide how much time he can invest and when and tag his tasks.

> I would ideally like all source files to use the same convention for line endings.

Good point! Let's do that. I created issue #31 and assigned it to you. Let's do it in V1.6. BUT: Let's do this as one of our very last actions we do, just before releasing. Otherwise this will be a merge nightmare for the three currently active branches.

sy2002, 2020-08-05:
### Update: Done regarding `dev-io-addr`:

- [x] Recompile standard C library
- [x] Perform some smoke tests for the recompiled standard C library
- [x] Recompile all /qbin demos and tools and check-in the new /qbin
- [x] Create a new disk image
- [x] Modify emulator/run_vga.bash to support the new disk image
- [x] Do not update the website http://qnice-fpga.com/emulator.html as nobody would notice the MMIO change there, BUT: We need to update the source code to download the V1.6 version of the disk image (with the old memory map, as the online demo of the emu is still working in the old world). Then: Do a smoke test locally by recompiling and testing the WebAssembly version of the emulator.

### What remains to be done regarding `dev-io-addr`:

- [ ] MEGA65: Update the hardware, update the test programs, retest on hardware
- [ ] Put the new disk image also to the two standard SD card (Nexys4DDR and MEGA65) and test them there
- [ ] Fix newly found bug while smoke testing recompiled qbins: emulator crashes on sierpinski.out (see new issue #32)

Looks like we are nearly there, when it comes to the topic of "Change the memory map and refactor MMIO registers"

sy2002, 2020-08-18:
CONSOLIDATION => CLOSED/DUPLICATE => Will continue to finish this in issue #69

closed 2020-08-18 by sy2002