sy2002/QNICE-FPGA #30: Tristate Refactoring

sy2002, 2020-08-05:
For 1.7 (or later): As described by Michael in issue #4 ISE vs. Vivado Riddle and #22 Refactor mmio_mux, the fact that we have this nice `inout` data bus plus the tristate buffers seem to be the reason for more stress and trouble, than it is good.

So we need to refactor this and I suggest that Michael is doing it in V1.7 or later using a new branch then which is a descendant of the then up-to-date `develop` branch.

MJoergen, 2020-08-15:
This is implemented in the separate branch `dev-remove-Z`.

Tested and verified using Vivado:
* Q-TRIS loads from SD-card and runs without problems.
* `ise.asm` and `eae.asm` work.

The constraint file `qnice_nexys.xdc` had to be updated because Vivado for some reason was renaming signals.

Next is to test using ISE, and examine Issue #4.

MJoergen, 2020-08-15:
I've now verified this change works with ISE and indeed DOES solve Issue #4!

So the question remains when to merge in these changes?

sy2002, 2020-08-15:
Hi Michael, HOORAY 😄 🥳   This is great :-)  Could you please do one more test, i.e. "timer_test.asm" on hardware where one part is being output on UART and one on VGA? This also tests some complex potential for bad behaviour on MMIO devices. If this also works, then please merge to `develop`.

sy2002, 2020-08-15:
And one more thing: Please sift through all the VHDL comments of all affected MMIO devices and correct the comment, that today says something like "data goes high impendance on disable" or something like this, so that we have consistent comments.

sy2002, 2020-08-15:
@MJoergen I do have another question for you in the context of this Tristate Refactoring: I do not get it why this works at all :astonished: - and still it does as you tested it. So here is my question: Each **inactive** MMIO device is putting `data_out` to `(others => '0')` and the one **active** MMIO device is putting data on the bus, for example `data_out <= x"1234"`. Why aren't we getting a "multiple drivers for one register" problem here? We are not "decoupling" the devices from the data bus, we just let all of them (but one) write zero on the data bus. And the one active writes x"1234". How does the Synthesizer know when to take the zero as the correct and true value and when to take x"1234" as the correct and true value? If 10 different devices are pulling at the register, how does it "know" which one is the right one?

MJoergen, 2020-08-15:
You are indeed correct, except for:

>  If 10 different devices are pulling at the register, how does it "know" which one is the right one?

You see, the devices are all pulling **separate** signals. I've added a bunch of new signals to the top level entity `env1.vhd`, e.g.
* `signal switch_data_out        : std_logic_vector(15 downto 0);`
* `signal kbd_data_out           : std_logic_vector(15 downto 0);`
* `signal timer_data_out         : std_logic_vector(15 downto 0);`

etc.

The data into the CPU is then generated by OR'ing together these individual signals, see lines 438-452 of `env1.vhd`.

MJoergen, 2020-08-15:
Verified with latest `develop` branch.

closed 2020-08-15 by MJoergen

sy2002, 2020-08-15:
Great Job, Michael! I also confirmed (using my favorite 3-way-multitasking test). Works great. And I am so glad to hear from you, that this also solved the ISE riddle ;-) I will also recheck the ISE thing on my side and then close #4.