sy2002/QNICE-FPGA #31: All VHDL files: Consistent line endings

sy2002, 2020-08-05:
Let's have consistent line endings in all VHDL files from version 1.6 on. But **CAUTION**: Let's do this change as one of the **last changes we do**, right before releasing. Reason: If we did it right now, we would have so many changes to soo many files that the process of merging `dev-io-addr` plus `dev-int` plus `develop` plus `master` would be a nightmare.

The reason why all these files are so scrambled, when it comes to line endings: I developed pre V1.6 with ISE under Windows. In V1.6 I switched to Vivado on Linux.

@MJoergen I guess Linux endings are fine, as all of us seem to run Vivado on Linux. I never did a "line ending change" in past and I assume you have some tool or something with which you could do that?

Part of release scope 1.6, issue #28 

Branch to execute this on: `develop` but only **after** `dev-io-addr` and `dev-int` are 100% done, tested and merged.

MJoergen, 2020-08-06:
I agree that we should be careful with the timing of this change. Just before release sounds reasonable.

MJoergen, 2020-08-18:
@sy2002 : I've updated the line endings now. So please do a "git pull" or similar, before proceeding e.g. with Issue #69.

closed 2020-08-18 by MJoergen