sy2002/QNICE-FPGA #32: Emulator crashes on EAE (?) in sierpinski.c

sy2002, 2020-08-05:
@bernd-ulmann Emulator crashes on EAE (?) in sierpinski.c in branch `dev-io-addr`. So this might be a problem related to the new MMIO address map.

Reproduction:

1. Make sure that you PULL the latest `dev-io-addr` branch so that you have the correct standard C lib (with the updated MMIO address map)
2. Make sure that you re-run `monitor/compile_and_distribute.sh` so that you have the correct/updated MMIO address map in dist_kit and that your local monitor.out is fresh
3. Go to the folder `c` and enter `source setenv.source`
4. Go to the folder `c/test_programs` and enter `qvc qvc sierpinski.c -c99`
5. Go to the `emulator` folder and enter `./make.bash`
6. Enter `/qnice`
7. Enter `load ../monitor/monitor.out`
8. Enter `load ../c/test_programs/sierpinski.out`
9. Enter `run 0`
10. Enter `C``R` `8000`
11. Enter `495,3,3`

You will receive a floating point exception.

```
TITANIUM-M5:emulator mirko$ ./qnice
Q> load load ../monitor/monitor.out
Unable to open file >>load<<
Q> load load ../monitor/monitor.out^C
Q> load ../monitor/monitor.out
Q> load ../c/test_programs/sierpinski.out
Q> run

Simple QNICE-monitor - Version 0.8 (Bernd Ulmann, sy2002, July 2020)
------------------------------------------------------------------------

QMON> CONTROL/RUN ADDRESS=8000
Sierpinski Fractal Generator
by LambdaBeta in August 2015, adjusted for QNICE by sy2002 in October 2016

Enter bitpattern, scale (2 .. 5) and generation count (0 .. 5).
Here are some value pairs that produce nice results:
 [b, s, g] = 495, 3, 3
 [b, s, g] = 7, 2, 5
 [b, s, g] = 186, 3, 3
 [b, s, g] = 495,3,3

Floating point exception: 8
```

sy2002, 2020-08-07:
Bernd found out (in German):

>kann es sein, dass der C-Compiler irgendwie noch eine falsche Memory-Map fuer
EAE nutzt? Zumindest spontan sieht der Emulator aus meiner Perspektive
correct aus - nicht zuletzt wird auf die EAE-Register immer mit einem
==-vergleich und nicht einem Range durch < und > zugegriffen, und die
EAE-Registeradressen in sysdef.asm sind auch korrekt.

> Es geht an Adresse 80F3 schief, wo sich das hier findet:

```
80F0 0D2C MOVE    R13 R11
80F1 1FAC ADD     0001 R11
80F3 0B72 MOVE    @R11 @R12++
80F4 0FAC MOVE    FF1F R11
80F6 0FAD MOVE    0001 @R11
```

R12 enthaelt an dieser Stelle den Wert 0xFF1C, was IO$EAE_CSR ist...

Means, somehow and somewhere in the C system, the old memory map is used, even though I recompiled the standard C library... Strange. I will assign this to me for further investigation.

sy2002, 2020-08-13:
@bernd-ulmann `0xFF1C` is the correct value for the `IO$EAE_CSR`, so the C library is using the new MMIO addresses. Still, we receive a `Floating point exception: 8`. I would say: No matter what any code does within the emulator: The emulator should never crash. If the EAE divides by zero, the emulator might want to halt the running program, do a register dump and allow debugging. But not crash. So please reinvestigate again.

bernd-ulmann, 2020-08-14:
The emulator no stops execution of the QNICE code in case of an attempt to divide by zero in the EAE as well as in the case  of an illegal EAE-opcode. In these cases the user is brought back to the Q> prompt.

sy2002, 2020-08-14:
Thank you for the (as usual) extremely fast and great fix. I will continue to track down this monster bug.

sy2002, 2020-08-15:
This was not only a problem in the emulator: It was also there on the hardware. The reason was, that the C compiler's code emitter had hardcoded EAE registers. Now everything works again, including the sierpinski.c that originally triggered this bug. Enjoy the image here:
[image]

closed 2020-08-15 by sy2002