sy2002/QNICE-FPGA #35: Feature Request: Add interrupt when VGA retraces specific scan line.

MJoergen, 2020-08-06:
### Changes to programming model

Add a register (IO memory address) that reads back the scan line currently being displayed.
Add a (same or separate?) register to store the value of a specific scan line.
Add registers to enable/disable interrupts when current scan line is equal to configured scan line.

This should probably be implemented AFTER Issue #17.

sy2002, 2020-08-06:
Good stuff! If you allow me to be hoggish and add all my wishes  😄  :

Horizontal sync interrupt (just as you described): Yes 👍 

Additionally: Vertical retrace / vertical sync interrupt would be great, too, because in game programming, you are using this to trigger [double buffering](https://web.fe.up.pt/~pfs/aulas/lcom2018/at/graphics_topics.html), which is nice for graphics programming.

So bottom line:

1. Yes, please implement exactly what you described above: This can be used for a multitude of great graphics effects including: [Rasterbars](https://codebase64.org/doku.php?id=base:demo_programming), showing more colors on the screen than the system allows (by quickly switching the palette when a raster interrupt hits), showing more sprites than the system allows, etc.

2. Add an addtional register(s) for the vertical retrace interrupt, so that we can trigger double buffered screen switches (reduce flicker in animations)

MJoergen, 2020-08-06:
Do we really need BOTH a vertical retrace interrupt AND a horizontal scan line interrupt ? I mean, both interrupts occur once each frame, it's just that the VSYNC interrupt occurs at the end of each frame, whereas the SCANLINE interrupt can occur at a configurable part of the screen. If you "just" want a VSYNC interrupt, then configure the SCANLINE to be 480 (or whatever is the value just after a frame). This could even be the default value, so just enabling SCANLINE interrupt will give an interrupt simultaneously with the VSYNC signal. Or am I misunderstanding something ?

sy2002, 2020-08-07:
Good point. Let me ponder it for a few days. My graphics demo programming times are a few years ago, so I need to re-think, if we'd really need two interrupts, or if only one, as you described in your comment is good enough. I feel that you might be very right and that the SCANLINE interrupt is good enough.

Independent from this discussion: Let's assume for now, we would only have the SCANLINE interrupt, I changed the name of the issue back to your proposed name :-)

What needs to work is, that the programmer is changing the registers inside the ISR, so that you have an arbitrary amount of SCANLINE interrupts. Here is a quote from Wikipedia:

>Such an interrupt provides a mechanism for graphics registers to be changed mid-frame, so they have different values above and below the interrupt point. This allows a single-color object to have multiple, horizontal color bands, for example, or for a sprite to be repositioned to give the illusion that there are more sprites than a system supports. The limitation is that changes only affect the portion of the display below the interrupt. They don't allow more colors or more sprites on a single scan line. 

Example:

1. Set the register so that the ISR is triggered at line 100.
2. ISR does some magic when 100 is reached and changes the register so that the ISR is triggered again at line 300.
3. Next time, at 300 it does even more magic and the ISR decides that it also needs a VSYNC and therefore sets it to 480

And at each of these interrupts, the ISR is reading some register to learn: Are we currently at the 100, 300 or 480 interrupt?

Background: With such a technique, you can trick the system to show more sprites than the system supports. You can also draw "C64 style rasterbars", you can have more colors, because you reset the palette for example. An endless wealth of pretty cool graphics effects :-)

And in the extreme case, you might even want to be able to have an ISR at each scanline - and as said: It is the ISR itself that sets up the next line where the next interrupt shall happen.

MJoergen, 2020-09-02:
Scan line interrupt is now implemented in hardware.

A small assembly program is written to demonstrate: `vga_int.asm`. This demo program uses pixel scrolling to make a text scroll smoothly across the screen, all controlled by interrupts.

The documentation has been updated too.

I'm closing this issue, since all the other VGA features are collected in Issue #17 .

closed 2020-09-02 by MJoergen

sy2002, 2020-09-02:
WOW! This is **awesome** 👍 It is such a pitty that I do not have FPGA hardware at hand right now! I cannot wait to see all of this! 😃 🚀 🎆