sy2002/QNICE-FPGA #36: Monitor built-in disassembler should convert relative addresses

MJoergen, 2020-08-06:
When disassembling code using the Monitor's built-in disassembler, relative addresses are shown as just that: relative.

I would like the disassembler to show the corresponding absolute address. For instance, the command "M S 0000 0060" currently shows:

```
0000 FFA0 RBRA    005A 1
...
005A FFA0 RBRA    0E88 1
005C 9FB8 AND     00FF R14 

```

The instruction at 0000 jumps to address 005C. Showing the relative address (005A) makes the output harder to read.

This applies to the instructions RBRA and RSUB. The instructions ABRA and ASUB already show the absolute address. Alternatively, the relative instructions could show both the offset AND the absolute address, to make the distinction more clear.

Tentatively assigned to V1.7 if you agree.

bernd-ulmann, 2020-08-06:
I will take care of this☺️

sy2002, 2020-08-06:
@bernd-ulmann As you are currently heavily time-contrained: No worries if this takes a while and if it goes to V1.7 instead of the current V1.6. But if you can do V1.6, it would be awesome :-)

In both cases: Please use the branch `develop` for that (and **not** `dev-int` or `dev-io-addr`).

bernd-ulmann, 2020-08-07:
Done. :-) A typical output looks now like this:

QMON> MEMORY/LOAD - ENTER ADDRESS/VALUE PAIRS, TERMINATE WITH CTRL-E
080000FFA00800100000080020FFB0080030FFFC080040F020080050F130
QMON> MEMORY/DISASSEMBLE START ADDRESS=8000 END ADDRESS=8010
8000 FFA0 RBRA    0000 1   (8002)
8002 FFB0 RSUB    FFFC 1   (8000)
8004 F020 RBRA    R0  1
8005 F130 RSUB    R1  1
8006 0000 MOVE    R0  R0

The absolute destination address is denoted in parenthesis following an RBRA/RSUB instruction.

sy2002, 2020-08-07:
@bernd-ulmann Thanks for this feature improvement. Can you please merge this it to `dev-int`? Background: It will make merging much easier later, because we have the the newest monitor version that was created when you teached the disassembler to learn the new interrupt opcodes in `dev-int`.

bernd-ulmann, 2020-08-07:
Done. :-)

closed 2020-08-07 by bernd-ulmann

sy2002, 2020-08-07:
@bernd-ulmann  and @MJoergen   Important hint: Always, when you change the monitor, then the following steps need to be done:

1. run `/monitor/compile_and_distribute.sh`
2. Read on the command line how large the rom is after the change. Example after Bernd's recent changes: `qasm2rom: 6498 ROM lines written.`
3. Go to `vhdl/env1_globals.vhd` to line 14 and change the variable

```
constant ROM_SIZE             : natural   := 6523;
```

So I did this now for this change Bernd did. New value is now: 6498  (strange: @bernd-ulmann  it shrinked due to the new feature ?)

I checked in the new file. Additionally also in the `vhdl/MEGA65` folder there is a `mega65_globals.vhd` which needs a similar change.

@MJoergen Maybe we can get smarter and get rid of this manual effort by improving the way the GENERIC of the `block_rom.vhd` and how it is instantiated...

sy2002, 2020-08-07:
oh: now after Bernd fixed his debug lib: ` 6548 ROM lines written.`

sy2002, 2020-08-07:
updated again

MJoergen, 2020-08-07:
> @MJoergen Maybe we can get smarter and get rid of this manual effort by improving the way the GENERIC of the block_rom.vhd and how it is instantiated...

I've created a new Issue #39 for this purpose.