sy2002/QNICE-FPGA #39: Remove generic ROM_LINES from block_rom.vhd and rom_from_file.vhd

MJoergen, 2020-08-07:
This will simplify the release process, by not having to update these values manually.

sy2002, 2020-08-08:
Cool - I did not even know, that this is possible ;-) Would be awesome, after you fixed that for the nexys4ddr hardware, if you could also fix it for the MEGA65 hardware and assign the issue to me afterwards so that I test on real MEGA65 hardware. If I got it correctly, that means that you would change the following files:

```
vhdl/env1_globals.vhd
vhdl/block_rom.vhd
vhdl/hw/nexys4ddr/env1.vhd
vhdl/hw/MEGA65/mega65_globals.vhd
vhdl/hw/MEGA65/MEGA65_ISE.vhd
vhdl/hw/MEGA65/MEGA65_Vivado.vhd
hw/xilinx/nexys4ddr/ISE/env1.xise
hw/xilinx/nexys4ddr/Vivado/qnice_nexys.xpr
```

This is quite a lot :-)

**Therefore: I propose to do that *after* we got rid of all the branches and are all back on `develop` but *before* we release 1.6? Or move it to 1.7? What are your thoughts?**  

### And here comes a slightly off-topic question for you (that we might want to consider for V1.7 - oder later?)

* One of the basic ideas of QNICE-FPGA is, that it is meant to be highly portable.  (For sure we are not there, yet ;-) Currently, everything is quite xilinx specific)
* Right now, we have two hardware targets (nexys4ddr and MEGA65).
* I assume that there will be more hardware targets in future, because for example the [Terasic DE10-Nano](https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&No=1046) dev board is becoming highly popular (and I recently bought one ;-)). 
* And then, the nexys4ddr is deprecated, the successor is called [Nexys4 A7](https://store.digilentinc.com/nexys-a7-fpga-trainer-board-recommended-for-ece-curriculum/]: I never tested on it, because I do not have one, yet. We might want to suuport the **Nexys A7** one day without giving up the Nexys4 DDR support.
* So the `vhdl/hw` as well as the `hw/xilinx` , `hw/intel` folders might grow in future, making maintenance hard

**Is there a way to work in VHDL a bit similar than with the C preproccessor? And then add some build system like CMAKE or the GNU build system?**

E.g. in the source files as high level as `#ifdef xilinx` and/or as low level as `#ifdef enable_hdmi`  and then something like

`make nexys4`
`make MEGA65`
`make DE10`

And the build environment is smart enough to grasp: OK, nexys is xilinx, DE10 is intel...

Can we unite the zoo of vhdl files that right now have been doubled just because I did not know better how to do it? Like uniting MEGA65 globals and env1 globals. Uniting MEGA65_ISE with MEGA65_Vivado with env1 ISE and Vivado and so on.

Having some `vhdl/hw/common` folder where all the common stuff resides and then the TRULY hardware specific stuff, such as HDMI and HyperRAM for MEGA65 then in a `vhdl/MEGA65´ folder.

As said: Slightly off-topic for now but maybe something for the future, if you have some thoughts how to tackle this...

MJoergen, 2020-08-08:
The initial intent with this issue was to make small easy changes. Definitely something that we can do just before release of V1.6, together with the line ending cleanup.

Your extra suggestions are definitely the direction we should go in, and indeed is very much possible and doable. But that is best postponed to V1.7. I think we have more than enough for V1.6 already. So perhaps make a new issue to increase portability and clean up the build environment.

sy2002, 2020-08-08:
You are right. :-) So I made an edit, deleted all the off-topic stuff and made a new issue #41 for the future stuff.

And this issue is now cleanly about the generic ROME_LINES. 

Let's do it in `develop` after we merged all the branches back to `develop` and as I said in the EDIT above: When doing this clean-up we need to take care of the `PORE` ROM, too.

MJoergen, 2020-08-14:
Fixed and verified using both ISE and Vivado.

Still need to deal with MEGA65 target => Reassigned to @sy2002.

sy2002, 2020-08-14:
Thank you. This is an important automation step and reduces the overall release and maintenane complexity. I will take care of the MEGA65 target.

sy2002, 2020-08-14:
Just a quick question on the implementation: The size of the ROM seems to be now constantly ~16kWords? Or is the synthesizer smarter and just will utilize as much space as we need?

MJoergen, 2020-08-14:
The synthesizer is smart and will make the same implementation as before.

closed 2020-08-15 by sy2002