sy2002/QNICE-FPGA #40: Bug in emulator: Overflow calculation during SUB is incorrect

MJoergen, 2020-08-08:
The following snippet of code:
```
MOVE    0x0005, R1
MOVE    0x0007, R2
SUB     R2, R1
```
calculates the subtraction 5-7 and gets the correct result -2 (represented as FFFE), but the overflow flag is set. In this calculation there is no overflow, and the overflow flag should be clear.

I assume the overflow flag should work just as on the 6502 processor. From the document [http://www.righto.com/2012/12/the-6502-overflow-flag-explained.html](http://www.righto.com/2012/12/the-6502-overflow-flag-explained.html) I have extracted this excerpt:

> The formulas for overflow during addition given earlier all work for subtraction, as long as the second argument (N) is ones-complemented. Since internall subtraction is just addition of the ones-complement, N can simply be replaced by 255-N in the formulas.

This could perhaps be used as part of the fix in the emulator.

bernd-ulmann, 2020-08-08:
Thank you so much for your work! I am really overwhelmed by what you already did! I will take care of this bug (and the X-flag) tonight when I will be back home. Have a great and sunny Saturday! :-)

bernd-ulmann, 2020-08-09:
Hi Michael - I am pretty confident that the overflow bit handling is now correct. Unfortunately, the cpu_test.asm HALTs at address 8394. It looks to me as it falls through the RBRA L_SUB_01, Z which seems OK to me as the comparison of R9 with R9 should not set the Z flag. Could you please have a look if that's due to a typo in the cpu_test.asm or if there is still something wrong with the condition bit handling in the emulator?

MJoergen, 2020-08-09:
I tried with the same code snippet again
```
MOVE    0x0005, R1
MOVE    0x0007, R2
SUB     R2, R1
```
and it still sets the overflow bit.

I think the overflow bit should be clear after this calculation. The calculation 5-7 = -2 should set the carry flag, but not the overflow flag.

bernd-ulmann, 2020-08-09:
Hmmmm... I reread http://www.righto.com/2012/12/the-6502-overflow-flag-explained.html and interpret the last table (just before "Overflow Myths") so that the subtraction 0x50 - 0xb0 should set C as well as V.

MJoergen, 2020-08-09:
I agree, that 0x50 - 0xb0 should set C as well as V. Reason is that 80 - (-80) = 160 is outside the range -128 to 127.

But what about line 3 of the same table: 0x50 - 0x70 ? There C is set (it's inverted on the 6502) and V is clear.

bernd-ulmann, 2020-08-09:
...good point - this implies that the formula given in the post, (M^result)&(N^result)&0x80, only holds true for addition and not subtraction. What would be correct for SUB?

MJoergen, 2020-08-09:
I think you should just invert N in the formula you cite.

bernd-ulmann, 2020-08-09:
Sounds reasonable. I changed it in the emulator. :-) (Fingers crossed that it works - sorry for the short replies, I am currently a bit overloaded by tasks, all of which are more "important" but also more boring than QNICE ;-) ).

MJoergen, 2020-08-09:
Well, I've verified and it works now. Closing this issue now.

closed 2020-08-09 by MJoergen