sy2002/QNICE-FPGA #43: Assembler Bug: Type leads to segfault

sy2002, 2020-08-09:
This code snippet leads to a segfault in the assembler:
```
        SYSCALL(puthex, 8)
```
Obviously, it does not like my typo `,8` :-)  The output looks like this:
```
TITANIUM-M5:test_programs mirko$ ../assembler/asm timer_test.asm 
assemble: Line 546: Illegal condition flag! (        ASUB puthex, 8)
../assembler/asm: line 36: 83225 Segmentation fault: 11  $assembler $temp_file $destination
```
**Branch for the fix: `dev-int` because there, the newest assembler (enhanced while adding the interrupt features) should be located there**

bernd-ulmann, 2020-08-09:
This error was due to not performing the memory allocation for the entry-data structure in case of an unrecognised condition flag. This has been fixed. :-)

closed 2020-08-09 by bernd-ulmann

sy2002, 2020-08-09:
Thank  you. Retested and confirmed.