sy2002/QNICE-FPGA #44: Emulator won't link on latest dev-io-addr branch

MJoergen, 2020-08-09:
Checking out the latest `dev-io-addr` branch and compiling the emulator leads to the following error:

```
timer.c:(.text+0x1aa): undefined reference to `pthread_create'
```

One possible fix to this is to replace (in `make.bash`) the line
`$COMPILER $FILES -O3 $DEF_SWITCHES $UNDEF_SWITCHES -o qnice
`
with
`$COMPILER $FILES -O3 $DEF_SWITCHES $UNDEF_SWITCHES -pthread -o qnice
`

Whether this is the correct solution (i.e. whether it works on other Linux distributions) I don't know. I'm using Linux Mint 19.3 (derived from Ubuntu).

sy2002, 2020-08-09:
Interessting - all these subtle differences between Linux and MacOS :-) On macOS it seems to work fine, I just checked it.

Since we are including our OS detection from `source ../tools/detect.include` in `emulator/make.bash`: Could you please for the `make.bash` and `make-vga.bash` add something like

```
if [ $OSTP = "LINUX" ]; then
    $MORE_SWITCHES = -pthread
fi;

$COMPILER $FILES -O3 $DEF_SWITCHES $UNDEF_SWITCHES $MORE_SWITCHES -o qnice
```

(Or if you find something more elegant)

And then also test, if `make-vga.bash` still works?

After that, please assign to me, because I need to look how we handle the pthread topic at the Emscripten / WebAssembly target anyway.

MJoergen, 2020-08-09:
No changes were necessary to make-vga.bash, because the SDL2 library automatically includes "-lpthread". So only the make.bash has been updated.

Reassigned to @sy2002.

sy2002, 2020-08-13:
The pthread based timer interrupt implementation is not working on the WebAssembly target.

closed 2020-08-13 by sy2002