sy2002/QNICE-FPGA #45: Timer Interrupt Hardware: Check Edge Cases and Stability

sy2002, 2020-08-09:
While implementing the timer interrupt hardware I noticed, that during the "Interrupt Grant Phase" when the CPU's `IGRANT_N` goes to `0` we need to de-couple all other devices from the DATA bus. Otherwise we might have a data clash on the bus.

Right now this logic is implemented in `mmio_mux` by making sure that all the devices, including RAM and ROM and PORE are not enabled (i.e. the devices `en = 0` and of course `we = 0`).

I did not fully think through all edge cases.

For example what about the whole topic of `cpu_wait_for_data` in `mmio_mux`. On a first glance and thought, it looks innocent, but at least `uart_cpu_ws` needs a second thought or in general the question: Will a disabled device automatically lead to correct behavuour in (a) all contexts and (b) in the particular context of `cpu_wait_for_data`?

**Are there more edge cases?**

**This topic has the potential for unpredictable crashes and nearly non-debuggable strange behaviour.**

I ran into it when I tried to bring the hardware timer interrupt to life on real hardware. While in the simulation, everything worked like a charm, on hardware, literally nothing worked.

Therefore I plan to leave this issue open for a few releases and go forward like this:

1. For V1.6: When everything is merged to `develop` and seems to work, rethink this and think about edge cases.

2. After releasing V1.6, do not close this but re-label it to `V1.7`. Rethink the edge cases again, also in the context of the tristate-refatoring in issue #30 . 

3. If possible for V1.7, write some timer interrupt stress test, that tries to provoke the edge cases, such as heaving timer activity while heavily working with the UART and it's `uart_cpu_ws` and similar with RAM/ROM and it's wait-for-data equivalents

4. If all is good in V1.7, too, then we can think about closing this issue - or - pushing it forward one more release, just not to forget about it

MJoergen, 2020-08-09:
Good catch. We definately need to be aware of this. It would be great with some tests to stress the system (both emulator and hardware).

sy2002, 2020-08-18:
Moved from V1.6 to V1.7 because all tests so far indicate that the interrupt system in general and the timer interrupt in particular are both reasonably stable: Everything seems to work great, and apart from the known nasty bug in issue #62 which is not a real-life application/situation and not easy reproduce, this is absolutely good enough to release.

sy2002, 2020-09-09:
This became obsolete with the timer adopting the new Daisy chaining module.

closed 2020-09-09 by sy2002