sy2002/QNICE-FPGA #46: Bug in emulator: SHR zero bits gives garbage in R14

MJoergen, 2020-08-09:
The following sequence of instructions:

```
MOVE    0x8765, R1
MOVE    0x00FF, R14
HALT ; Now X correctly contains 1
SHR     0x0001, R1
HALT ; Now X correctly contains 1
SHR     0x0000, R1
HALT ; Now X still contains 1, but should be cleared
```

leads to X not being cleared as it should. Or should it be unchanged?

Furthermore, the R14 register contains garbage as seen from the following run. It therefore looks like there is an uninitialized variable when shifting by a zero amount. Perhaps the call to `write_register()` should not take place in this situation?

```
Q> rdump
Register dump: BANK = 00, SR = MIVNZCX1

	R00-R03: 879f 8765 0000 0000 
	R04-R07: 0000 0000 0000 0000 
	R08-R11: 8000 0be1 0000 0000 
	R12-R15: 0000 feeb 00ff 8005 

Q> run 0x8005
HALT instruction executed at address 8007.

Q> rdump
Register dump: BANK = 00, SR = MIVNZCX1

	R00-R03: 879f c3b2 0000 0000 
	R04-R07: 0000 0000 0000 0000 
	R08-R11: 8000 0be1 0000 0000 
	R12-R15: 0000 feeb 00ff 8008 

Q> run 0x8008
HALT instruction executed at address 800A.

Q> rdump
Register dump: BANK = 170f12, SR = MIVNZC_1

	R00-R03: 0000 c3b2 0000 0000 
	R04-R07: 0000 0000 0000 0000 
	R08-R11: 8000 0be1 0000 0000 
	R12-R15: 0000 feeb 170f12fd 800b 
```

bernd-ulmann, 2020-08-09:
Phew, that is an interesting observation. Now that I think of it it does not seem too unreasonable that a shift by zero places leaves the status register bits untouched.

What is your opinion on that?

MJoergen, 2020-08-10:
Hmmm. I agree that shifting by zero should leave the status register bits untouched. That seems like the most reasonable thing to expect.

But there is still something fishy with the R14 register when shifting zero:
```
Q> rdump
Register dump: BANK = 170f12, SR = MIVNZC_1

	R00-R03: 0000 c3b2 0000 0000 
	R04-R07: 0000 0000 0000 0000 
	R08-R11: 8000 0be1 0000 0000 
	R12-R15: 0000 feeb 170f12fd 800b 
```

bernd-ulmann, 2020-08-10:
Could you please try again, I changed read_register(...) slighty to ensure that all values returned are only 16 bit long. :-)

MJoergen, 2020-08-10:
> Could you please try again, I changed read_register(...) slighty to ensure that all values returned are only 16 bit long. :-)

Have you push'ed your changes? The github repo has no changes to `read_register()`.

bernd-ulmann, 2020-08-10:
Ups... I pushed but the push failed which I did not see... I have an inconsistency. Unfortunately I have a phone conference starting in 3 minutes. I will do it afterwards (at about 1:00 PM).

MJoergen, 2020-08-10:
I've updated cpu_test.asm, and added some comments, so the code should be more readable.

The current repo (HEAD at dev-io-addr) fails when SHL'ing zero bits: The carry bit gets cleared. This is in line 1036 of cpu_test.asm.

bernd-ulmann, 2020-08-10:
AAARGL! I will take care of that...

bernd-ulmann, 2020-08-10:
Phew! Your wonderful CPU test suite now finishes with "OK"! :-)

closed 2020-08-10 by bernd-ulmann