sy2002/QNICE-FPGA #47: Feel the joy: Hardware Timer Interrupt

sy2002, 2020-08-09:
Gentlemen, this is an unusual "issue". It is just meant to share some joy with you about the first (still work-in-progress) version of the hardware timer interrupt in branch `dev-int`: This branch now successfully runs `test_programs/test_int.asm` (software interrupts) and `test_programs/timer_test.asm` (hardware timer interrupts). But this is by far not all ;-) Read on!

IT IS SOOOOOO FREAKING AWESOME TO SEE. 😃 

Here is what I did:

1. Synthesized the Nexys 4 DDR Version in `dev-int` and start it on the FPGA
2. STDIN/STDOUT => UART (SW0 and SW1 are OFF)
3. Assemble `test_programs/timer_test.asm`
4. Connect via terminal program to UART and `M` `L` paste the timer test `.out` into the terminal window and start it via `C` `R` `E000`

The output that then comes on serial is:

```
QMON> CONTROL/RUN ADDRESS=E000
Setup timer 0 to interrupt every 1000 milliseconds.

QMON> Timer 0 has issued interrupt request #0001
Timer 0 has issued interrupt request #0002
Timer 0 has issued interrupt request #0003
Timer 0 has issued interrupt request #0004
Timer 0 has issued interrupt request #0005
```

This is already pretty cool. But hang on, it gets cooler: Of course, the monitor is still working fine, as we are talking about real hardware timer interrupts. So even if the ISR spills our terminal window, the `QMON>` prompt still works.

So - in my Nexys board, I have a FAT32 formatted SD Card, so I entered, while the ISR continues "spilling" and actually interrupting my while I type :joy: `F` `R` `qbin/q-tris.out` as you can see here

```
Timer 0 has issued interrupt request #0013
Timer 0 has issued interrupt request #0014
FILE/Timer 0 has issued interrupt request #0015
RUN: qbTimer 0 has issued interrupt request #0016
in/q-tTimer 0 has issued interrupt request #0017
ris.outTimer 0 has issued interrupt request #0018

Timer 0 has issued interrupt request #0019
Successfully loaded qbin/q-tris.out from 8000 to 91E3
Running...
Timer 0 has issued interrupt request #001A
Timer 0 has issued interrupt request #001B
Timer 0 has issued interrupt request #001C
```

(The ISR sits above `E000` and Q-TRIS loads to `8000` to `91E3`, so no overlap.)

And then - I can play Q-TRIS on the VGA screen, while on the terminal, it keeps counting. Slowly and steadily. Kind of a "multi monitor setup": VGA = Q-TRIS and UART = show ISR output. And of course you do not feel any disturbance because with our ~12 MIPS @ 50 MHz, QNICE is surely fast enough for that.

**What a joy** 😄 

And then, in the end, the whole thing is deactivated by running `C` `R` `E100` as seen here:

```
Timer 0 has issued interrupt request #007C
Timer 0 has issued interrupt request #007D

QMON> Timer 0 has issued interrupt request #007E
Timer 0 has issued interrupt request #007F
Timer 0 has issued interrupt request #0080
CONTROL/Timer 0 has issued interrupt request #0081
RUN ADDRESS=ETimer 0 has issued interrupt request #0082
100

QMON> 
```

As soon as both of you have seen/read this - one of you can close the issue ;-)

MJoergen, 2020-08-10:
Read and enjoyed !!!

sy2002, 2020-08-10:
Here comes a second round of joy: After having fixed tha combinatorial loop (thanks to Michael), now we have two hardware timer interrupt generators. As a pretty cool test, I enhanced `test_progrems/timer-test.asm` in a way, that it draw a "bouncing ball" (using the letter "O") with 25 Hz at the very first line of the screen. So my little "stress test" for both timers and therefore also for the **Daisy Chain Mechanism** was: The "two monitor setup" that is described above PLUS while I play Q-TRIS, the "bouncing ball" moves smoothly from left to right and right to left... :-) 

That means: I would say that the hardware implemention of the interrupt system is done (with the exception of a test of an edge case in the Daisy chain, but this has time and can be done after merging).

And that means: I will start merging `dev-int` to `develop`, soon.

closed 2020-08-12 by sy2002