sy2002/QNICE-FPGA #49: CPU R14 (status register): Mismatch between hardware and emulator

MJoergen, 2020-08-10:
The instruction
```
MOVE 0x00FF, R14
```
works differently on the emulator and the hardware.

On the emulator, all flags (in particular the overflow flag V) are set.

On the hardware, the overflow flag is not set.

In the file `qnice_cpu.vhd` in lines 580-587 I see the comment:
```
-- not all parts of the SR are writeable: only the upper 8 bit plus
-- the M, C and X register are writeable, 
```
So what is the 'correct' behaviour? And what is the 'M' flag ? I don't see it mentioned anywhere in the QNICE introduction document.

sy2002, 2020-08-10:
@bernd-ulmann Question for the ISA father: for some reason back in the days in 2015 I wrote the following comment:
```
-- not all parts of the SR are writeable: only the upper 8 bit plus
-- the M, C and X register are writeable, 
```
Well - we just killed `M` the other day, but is it still true that only C and X is writeable?
I do not remember, why I wrote that comment and why I implemented it that way - maybe you have some thoughts?

bernd-ulmann, 2020-08-10:
The M (and I) bits have been buried in the mean time. These were originally intended for interrupt handling (a mask bit, enabling/disabling interrupts and an I-flag that should have denoted that an ISR is currently executed). 

Although the incredible heat makes thinking rather difficult at the moment, all of the SR-bits should be writable so that they can be set/cleared with a MOVE, AND, OR, XOR.

sy2002, 2020-08-10:
OK, understood for all the flags that are currently in use. But would you also  - for now, while we do not have any use for the buried M and I flags - allow these two "reserved" flags to be modified or shouldn't we force them to zero until we make use of them in future?

sy2002, 2020-08-10:
Bernd just answered my last question outside this issue via email. Answer: SR should be fully writeable, just like it is done in the emulator. So I assigned this issue to me and will change the hardware accordingly.

MJoergen, 2020-08-11:
Ok, so I was impatient (!) and I tried changing the lines

```
-- not all parts of the SR are writeable: only the upper 8 bit plus
-- the M, C and X register are writeable, 
fsmSR(15 downto 8) <= std_logic_vector(Alu_Result(15 downto 8));
fsmSR(7) <= std_logic(Alu_Result(7)); -- M
fsmSR(2) <= std_logic(Alu_Result(2)); -- C
fsmSR(1) <= std_logic(Alu_Result(1)); -- X
```

to this single line:

```
fsmSR <= std_logic_vector(Alu_Result);
```

I thought this would solve the problem. However, the program `cpu_test.asm` still fails in hardware at address `0x8015`.

So I hooked up the Internal Logic Analyzer and got the following screen shot:
[image]

This image corresponds to the following sequence of instructions:
```
8007  FFA0  0006  RBRA L_UNC_3, 1
800F  0FB8  00FF  MOVE 0x00FF, R14
8011  FFAD  0003  RBRA E_R14_ST_01, !V
8013  FFA5  0002  RBRA L_R14_ST_01, V
8015  E000        HALT
```
This shows that after executing the instruction `MOVE 0x00FF, R14` the Status Register correctly contains `0x00FF`. Then the processor executes `RBRA E_R14_ST_01, !V` and (correctly) does not take the branch. However, the Status Register gets cleared during this instruction, and then the following branch is not taken either, leading to the HALT instruction.

Clearing the Status Register is unexpected, but in line 532 I see the following
```
fsmSR <= SR(15 downto 8) & "00000001"; -- clear flags
```
This is not the same behaviour as the emulator.

I'm not sure how to proceed from here, so I'll leave it assigned to @sy2002 .

sy2002, 2020-08-11:
@MJoergen WOW - I can absolutely not remember why I did that back in the days... 😳     From today's perspective it looks like one could just comment this out and everything works? But I fear, that - as always in Hardware development - it is not that easy?  I will look into it tonight. If you keep feeling impatient, feel free to test, if commenting out this line would help 😉 But please be aware, that the newest CPU is in the branch `dev-int`, so please do not check in changes to the CPU in `dev-io-addr`. If the fix works: Feel free to check it in in `dev-int`.

(Off topic, but this ILA seems to be *really* cool! Can you show me at some videocall/screen sharing session how to use it?)

MJoergen, 2020-08-11:
Regarding ILA, it is usually much easier to debug in simulation rather than using ILA. But yeah, ILA is very useful in some situations.

I normally use Vivado in non-project mode and build everything (including ILA) from the command line using tcl scripts. But I'm getting comfortable now using the GUI and project-mode to play with ILA and get useful results that way.

For sure I'll show you how to use the ILA. It's essentially like a digital oscilloscope. Just let me know when a convenient time is for you. It should take an hour or so, but good to reserve two hours just in case.

I tried your suggestion about removing the line that clears the status flags, and now the test proceeds until `0x8390`, which is label `E_MOVE_CV_11` in `cpu_test.asm`. Essentially, this test verifies that the MOVE instruction ONLY touches the Z, X, and N flags, and leaves the C and V flags untouched. It seems like the MOVE instruction modifies all status flags, which is not the same behavior as the emulator.

Since, as you mention, the hardware CPU is being developed in a different branch, perhaps we should leave any further hardware tests until we have merged everything together in the `develop` branch. In any case, I'm still working on the CPU test suite using the emulator, and I still have a lot to do on that part.

sy2002, 2020-08-11:
I just opened issue #51  "Make Hardware pass cpu_test.asm` and assigned it to me.

But let's leave this issue open, too, as I assume the fix will be rather easy after having read Michaels latest comment: So I would close this one, as soon as the original SR issue works and then we continue all cpu_test related things in issue #51

sy2002, 2020-08-15:
I close this one as DUPLICATE of issue #51 and will work at this topic there.

closed 2020-08-15 by sy2002