sy2002/QNICE-FPGA #50: Bug in emulator: CMP instruction modifies the wrong status bits

MJoergen, 2020-08-11:
This snippet from the emulator:
```
      if (cmp_0 > cmp_1)
        sr_bits |= 0x0020;  // Set C-bit
      else
        sr_bits &= 0xfffb;  // Clear C-bit
```
leaves me confused, because `0x0020` is not the inverse of `0xfffb`. This is comparing two signed numbers, so should that result be in the C bit or the N bit of the status register ?

bernd-ulmann, 2020-08-11:
That is really puzzling. I am currently not sure if the C- or V-flag should be affected - in either case one of the two bit masks is erroneous. To be honest, I have no idea why I implement CMP so differently from SUB. I just changed the code so that effectively a SUB is performed without the result being written back.

Regardless of this change, the cpu_test now HALTs at address 899F (I tested before and after this change).

MJoergen, 2020-08-11:
I've now updated the test with the new semantics of the CMP instruction and everything passes.

I did change the ordering of the arguments to CMP, to match the same ordering used in SUB. I hope that is ok? If so, then this issue can be closed.

bernd-ulmann, 2020-08-11:
That sounds great! Thank you very much! :-)

closed 2020-08-11 by bernd-ulmann