sy2002/QNICE-FPGA #51: Make the hardware pass `cpu_test.asm`

sy2002, 2020-08-11:
As soon as @MJoergen is ready with `test_programs/cpu_test.asm?`, @sy2002 will work on making the hardware pass the test. We will start with this, as soon as both branches `dev-int` and `dev-io-addr` are merged back to `develop`, so that this whole thing will be happening on `develop`.

MJoergen, 2020-08-11:
And let's hope we haven't broken anything else along the say. I.e. we should check all the other example programs, including Q-TRIS, etc.

sy2002, 2020-08-11:
Absolutely :-) The good old smoke test, that includes using the rather complicated FAT32 routines to load something from the SD card and then play a round of Q-TRIS while running two independent timers including a bouncing ball: If all that works, we may be confident 😃

sy2002, 2020-08-15:
First bugs: It halts at `0x8016` because the SR is not writeable. When fixing this, I also (as a first try) commented out the reset of the SR during each branch instruction. Now the cpu test runs until `0x8390`. I also did The Smoke Test, so I guess that the fix is OK.

sy2002, 2020-08-15:
The `cs_excute` stage of the CPU was updated to prevent the `C` and `V` flag from being changed during a MOVE operation. This lead to progressing the cpu test from the above-mentioned `0x8390` to now `0x85AE`, where the next `HALT` is being reached. I did The Smoke Test with these changes, seems to be all good.

MJoergen, 2020-08-16:
I now see the `cpu_test.asm` fails in hardware at address 0x844E. This is because the INCRB/DECRB test has been added.

The test fails after this sequence of instruction:

```
8414  0FB8  0000  MOVE 0x0000, R14 ; All flags initially clear
8416  E0C0        INCRB
8417  CFB8  0101  CMP 0x0101, R14 ; Verify flags unchanged
8419  FFAB  0033  RBRA E_RB_R14_01, !Z
```

The funny thing is that the status register R14 indeed does get the value 0x0101, yet the CMP instruction does not set the Z flag, see the screen shot below.

[image]

MJoergen, 2020-08-16:
Here is another screen shot with some more debug signals:

[image]

The yellow cursor is placed at the start of the `CMP 0x0101, R14` instruction, where ADDR = 0x8417.

The CPU state cycles through the four states:
- cs_fetch
- cs_decode
- cs_exeprep_get_src_indirect
- cs_execute

I don't know how to proceed from here, so I'll leave this with @sy2002 .

sy2002, 2020-08-17:
Thank you Michael for this exemplary debugging described in your comments above! The signal flows shown in your screenshots were so telling, that this helped me to fix the issue that you described, without even needing to fire up a simulation or something else. Great teamwork! Cool! 😃 

So [the issue you described](https://github.com/sy2002/QNICE-FPGA/issues/51#issuecomment-674520560) is fixed now:

**The cpu test is now able to progress further and is now stopping at address `0x85EA`**

I performed "The Smoke Test" to be sure that this fix is not destroying something very obvious: PASSED.

Things are progressing fine ;-) I will continue with "Make the hardware pass pu_test.asm" today later evening.

sy2002, 2020-08-17:
Fixed the SHR/SHL bugs in the CPU.

@MJoergen I added 8 more test cases to check SHL 0,0 and SHR 0,0  to make sure that no zero flag is being set and checked in the new version of cpu_test.asm => please check, if I did it correctly

sy2002, 2020-08-17:
Fixed the bug at `SWAP` at `0x88bb` which was due to `SWAP` touching C or V.

Did a retest of "The Smoke Test": PASSED.

It not stops at `0x88f9` which is the `NOT` command.

sy2002, 2020-08-17:
There is progress. The cpu test now runs until `8CE3`, which is `E_MOVE_AM_61`.
"The Smoke Test": PASSED.

sy2002, 2020-08-17:
Fixed the `E_MOVE_AM_61` bug, which was an interessting one (see commit).
"The Smoke Test": PASSED.
Now the cpu test runs until `0x8E46`, which is `E_MOVE_AM2_61`.

sy2002, 2020-08-17:
Fixed the `E_MOVE_AM2_61` bug.
"The Smoke Test": PASSED.
Now the cpu test runs until `93D5`, which is `E_SUB_AM2_071 HALT`.

sy2002, 2020-08-18:
Fixed the `E_SUB_AM2_071` bug.
"The Smoke Test": PASSED.
Now the cpu test runs until `9455`

sy2002, 2020-08-18:
@MJoergen and @bernd-ulmann Another reason to celebrate: **The hardware passes cpu_test.asm** 🎉 🚀 👍 😄 
And it passes "The Smoke Test" (i.e. 3-way multitasking incl. FAT32 and Q-TRIS).
@MJoergen WOW: I am very impressed by the **ingenuity** you have put into cpu_test.asm. My deepest respect!

So I am closing this issue now.

In case issue #68 leads to a change of the ISA and the emulator and cpu_test.asm, then this issue need to be re-opened.

MJoergen, 2020-08-18:
This is FANTASTIC!!!

Great work, both of you @bernd-ulmann and @sy2002 . This has been a real marathon-issue, with decisions to be made; new decisions changing old decisions; writing test code; writing more test code; updating emulator (several times); AND fixing quite a few (minor) bugs in the CPU.

When I started writing the `cpu_test.asm`, I didn't really expect it to find any bugs .....

I just want to congratulate you (well, all of us) for a great team effort!!!

I think the effort has been well worth it, and we are in a much better position going forward.

sy2002, 2020-08-18:
Reopened due to the ISA change in issue #68

sy2002, 2020-08-18:
The hardware CPU now implements the ISA change that SHL/SHR are now setting the N and Z flags (issue #68).
* Newest version of `cpu_test.asm`: PASSED
* "The Smoke Test": PASSED

closed 2020-08-18 by sy2002