sy2002/QNICE-FPGA #52: Perform Smoke Tests on merged branch develop

sy2002, 2020-08-11:
Gentlemen, I did it!! I merged `dev-int` and `dev-io-addr` into `develop` this evening!  🥳  🎉  :bowtie: 

There was astonishingly little trouble in the merging itself, because both of you were so disciplined, when committing to the different branches. So thank you very much for that!

**That means we do not need the branches `dev-int` and `dev-io-addr` any more** .

I am pretty sure, that we do not have merge errors, i.e. wrong versions or so. I double checked each file and when in doubt I used line-by-line merge and comparison tools such as Araxis Merge.

But I guess we all need to perform some smoke tests on `develop` now - also and in particular, because I did all this work at a place, where I don't have an FPGA at hand, so all I could test was on the level of the emulator.

Michael, if you ever wanted to reproduce this cool "dual monitor" setup including a "jumping ball" on top of Q-TRIS on your own hardware: This might be the time, here is the comment that I mean:

https://github.com/sy2002/QNICE-FPGA/issues/47#issuecomment-671589602

BUT - as always, when so many changes have been done: Something broke. Not because of the merge! But because of the change in behaviour of the emulator. Since this has nothing todo with the merge itself, I opened a new issue for that called "Recent emulatopr changes broke SD Card, FAT32 and QNICE". Let's use the new issue (issue #<todo>) to discuss and work at that.

Let's use this issue at hand for all other smoke test results on the merging itself. I assume the merging itself worked pretty fine and we will be able to close this issue rather sooner than later. But I feat that issue #<todo> will be with us for a little longer.

There is a question to Michael about the old branches: So far, I never dared to delete any old branch due to the fear of loosing the history and due to the fact that I do not know how to do it right. Do you have experience with GitHub's behaviour about that in conjunction with the underlying Git? Bernd and I do not. So if you do: We might want to discuss also the deletion of old branches here in this issue before closing. Otherwise, let us just let them be for a little longer until we dare to proceed :-)

sy2002, 2020-08-12:
Michael did a smoke test on hardware: Works as described here: https://github.com/sy2002/QNICE-FPGA/issues/53#issuecomment-673074943

sy2002, 2020-08-13:
I performed some more smoke tests on the software stack, taking the newest emulator into consideration as described in this comment  https://github.com/sy2002/QNICE-FPGA/issues/53#issuecomment-673674143 as this will be the basis for V1.6.

So we can consider the merge of `dev-int` and `dev-io-addr` into `develop` as successful.

The last remaining topic of this issue is this one:

> There is a question to Michael about the old branches: So far, I never dared to delete any old branch due to the fear of loosing the history and due to the fact that I do not know how to do it [...]

I am therefore assigning to Michael: If you have insights, please share. Otherwise, please just close this issue and we leave the old branches as they are for the time being.

MJoergen, 2020-08-14:
DISCLAIMER: I'm still learning git myself :-D I still occasionally mess up, and have to re-clone a repository, because I can't figure out how to clean up my mess...

Anyhow, what I know is that branches are just pointers (i.e. links). Deleting the branch itself does not delete the history. It just removes a convenient link to a given commit. So the question is, do we need to *refer* to these branches again, i.e. to point to the latest commit in that branch? Probably not.

For reference: One can always refer to previous commits using the hash, i.e. `aa6481e` (dated August 11) is the hash for `dev-io-addr`, and `1f2de4c` (dated August 10) is the hash for `dev-int`.

--------------------------

I did a bit of google'ing and found this nice explanation: [https://ardalis.com/why-delete-old-git-branches/](https://ardalis.com/why-delete-old-git-branches/)

A note of warning: In this link they run this command:
```
git branch --merged
```
to get a list of branches that have been merged. I just want to point out, that the output from this command is *relative to* the currently checked out branch, something that was not clear from the text in the link.

So on the current repo on my machine I get the following output:

```
$ git checkout master
Switched to branch 'master'
Your branch is up to date with 'origin/master'
$ git branch --merged
* master

$ git checkout develop
Switched to branch 'develop'
Your branch is up to date with 'origin/develop'.
$ git branch --merged
  dev-int
  dev-io-addr
* develop
```

So this indeed confirms that `dev-int` and `dev-io-addr` can safely be deleted, as expected.

sy2002, 2020-08-15:
OK I did it: Cleaned up our branch zoo! :godmode: 
The only remaining branches are:
```
master
develop            <== all regular development work goes here
dev-cpu-bus-test   <== Michael's 
gh-pages           <== Source code of http://qnice-fpga.com/
```
You might want to run this command to clean up your local repository:
```
git fetch -p
git branch -d <name of any local branch that is not there on GitHub any more>
```

closed 2020-08-15 by sy2002