sy2002/QNICE-FPGA #54: Second timer (aka timer1) does not seem to work in the emulator

sy2002, 2020-08-12:
How to reproduce:

* Experiment 1: run `test_programs/timer_test.asm` (which works on the hardware) on the emulator: Neither timer0 nor timer1 will start

* Experiment 2: comment out the invokation of timer1 (by not having any ISR written) and rename the timer0 registers to timer1 => timer1 will not start

bernd-ulmann, 2020-08-15:
The timer implementation now works with the simple_timer_test.asm.

@sy2002 : I found the problem of code being executed from "random" addresses in the emulator. I had a pointer error in the timer thread.

It still fails with the more complex timer_test.asm as a thread can not be canceled. My current assumption is that timer-start and -stop follow so quickly that the newly created thread is not in a state where it could be canceled. I will look into this but if one does not create and destroy timers in very rapid succession, everything seems to work.

sy2002, 2020-08-15:
Bernd, have you tried to comment out "the nasty block" in `test_program/timer_test.asm` which does this rapid sucession and then use the VGA emulator and look if the UART is filled with the output and the bouncing ball works?

bernd-ulmann, 2020-08-15:
No, but I can recreate the problem with the simple_timer_test.asm program. :-) I am working on it, now that both timers work in the emulator. :-)

sy2002, 2020-08-15:
Awesome :-) I am curious: What was the pointer error in the timer thread exactly about?

bernd-ulmann, 2020-08-15:
I used garbage as interrupt address in timer_1 as I forgot to multiply the id with the number of registers per timer... *ARGL* Incredibly stupid... :-}

sy2002, 2020-08-15:
Don't worry Bernd, it happens to the best of us! :-)

bernd-ulmann, 2020-08-15:
I have changed your (@sy2002 's) timer_test.asm and jump over the nasty register test which is not handled correctly at the moment as the timer module tries to cancel a thread which has not fully been instantiated by now. 

Without this, the test works like a charm, I see the "flying" character on the VGA screen and get the other timer's interrupt messages on the serial console. :-)

sy2002, 2020-08-15:
Cool, then you might want to close this issue one while opening a new which is tagged by V1.7 which reads something like: "Make the emulator's timer robust to handle nasty test programs :-)" (or something like that) 😄

closed 2020-08-15 by bernd-ulmann