sy2002/QNICE-FPGA #59: ISE target does not build with latest develop branch

MJoergen, 2020-08-14:
Checking out the latest develop branch and compiling using ISE leads to the following error:

```
ERROR:NgdBuild:604 - logical block 'timer_interrupt' with type 'timer_module'
   could not be resolved. A pin name misspelling can cause this, a missing edif
   or ngc file, case mismatch between the block name and the edif or ngc file
   name, or the misspelling of a type name. Symbol 'timer_module' is not
   supported in target 'artix7'.
```

Perhaps a file needs to be added to the project ?

Compiling using Vivado works fine.

sy2002, 2020-08-14:
Thank you for finding this one. I fixed it. Indeed these two files were missing:

https://github.com/sy2002/QNICE-FPGA/blob/develop/vhdl/timer_module.vhd
https://github.com/sy2002/QNICE-FPGA/blob/develop/vhdl/timer.vhd

I quickly made the following smoke test:

Synthesize and run `test_programs/int_test.asm` and `test_programs/timer_test.asm`: Both work fine: I used UART in conjuntion with VGA, so I had the timer messages ticking on UART and the "bouncing ball" running on VGA.

But when I then tried to load Q-TRIS, I had this result:

1. 7 Digit Display only shows 2 digits instead of 4 times `0` as usual
2. No Q-TRIS is starting.

What I did not test (because I am in a hurry): Is maybe also the Vivado bitstream not working any more? Or is this an ISE specific riddle, just as #4 ?

@MJoergen The rest of the day I have only my laptop with my, no FPGA... If you want to test your ILA, this might be a case for it.  But I am getting the gut feeling, that we might need to say goodbye to ISE - OR - do this trisate refactoring earlier than later. I would tend to "earlier" if you feel that you could do it (e.g. while I am on vacation).

sy2002, 2020-08-14:
oh sorry - I know why Q-TRIS did not work: on my SD card there is still the pre MMIO change version of Q-TRIS. But I need to run now. Maybe you want to retest this one with ISE and timer_test.asm and Q-TRIS in parallel? Otherwise I'll do it tomorrow 😃

MJoergen, 2020-08-14:
Verified successful build and that Q-TRIS works. Have not tested with `timer_test.asm`.

Closing this issue now as the main problem is fixed.

closed 2020-08-14 by MJoergen