sy2002/QNICE-FPGA #64: Update all documentation for 1.6

sy2002, 2020-08-15:
Update all documentation for 1.6

sy2002, 2020-08-15:
Convert (and maybe reformat) all *.txt documentation in /doc into *.md due to nicer readability within GitHub

sy2002, 2020-08-18:
Sift through all documentation parts (such as doc/README.md and others) and fix old MMIO addresses

Check and possibly fix the inital walkthrough

Add more TODOs from VERSIONS.txt here

sy2002, 2020-08-18:
@bernd-ulmann Currently, only the programming card explains the flag behaviour of CMP. The intro document does not.

bernd-ulmann, 2020-08-18:
I removed the SHL/SHR footnotes from both documents and added slide describing the behaviour of the CMP-instruction in the introductory slide set. :-)

sy2002, 2020-08-18:
@bernd-ulmann Sometimes I can be a nitpicker, I know, but I cannot stop it :-) Please look at this image, it shows the PDF version of the newly updated QNICE intro pdf:
[image]

sy2002, 2020-08-18:
as part of this issue: update doc/how-to-release.md

bernd-ulmann, 2020-08-18:
WOW! Eagle eyed! :-) It's fixed now. :-)

bernd-ulmann, 2020-08-18:
> as part of this issue: update doc/how-to-release.md and VERSIONS.txt

...I don't have these two files in doc/?!

sy2002, 2020-08-18:
if VASM is not updated in V1.6 (see issue #65) then we need to document this fact in constraints.txt

sy2002, 2020-08-18:
> ...I don't have these two files in doc/?!

@bernd-ulmann Sorry, I did not commit yet doc/how-to-release.md. Done now. VERSIONS.txt is sitting in the main folder.

sy2002, 2020-08-18:
In the QNICE main folder, there sits VERSIONS.txt. I rececntly updated it to reflect everything we did in V1.6 and boy is this a looooong list :-) I am pretty happy with the sheer volume of V1.6 ;-)

@bernd-ulmann Please look at all the @bernd tags in VERSIONS.txt and check, if I got everything related to your work in V1.6 right. Feel free to add/edit at will and then remove all @bernd tags.

@MJoergen Please look at the @michael tags in VERSION.txt and check, if I got everything related to your work in V1.6 right. Feel free to add/edit at will and then remove all @michael tags.

MJoergen, 2020-08-18:
I've checked my part in VERSIONS.txt and updated it.

sy2002, 2020-08-18:
I updated quite a lot of documentation. Here are the remaining TODOs for this issue:

- [ ] Sift through all documentation parts (such as doc/README.md and others) and fix old MMIO addresses
- [ ] Check and possibly fix the inital walkthrough
- [ ] After the bitstreams have been made and put to `dist_kit`: Fix the references in `hw/README.md` and `README.md`
- [ ] Update left column of Homepage http://qnice-fpga.com/ from "V1.5" to "V1.6" and change the ISE language to Vivado there and add Michael to the list of contributors there
- [ ] Finalize first version of `doc/how-to-release.md`

sy2002, 2020-08-18:
@MJoergen Can you have a look at the first two checkboxes in the list above? I would then take the others.

MJoergen, 2020-08-19:
* The top-level `README.md` refers to `dist_kit/qnice-v141.bit` in several places. I guess we really need two different bit-files: One for Nexys4DDR and one for MEGA65. @sy2002 
* The same file also refers to `dist_kit/q-tris.bit` in several places. The same comment applies here. @sy2002 
* The programming card needs to have `ADD/SUB 0x0100, R14` replaced with `INCRB/DECRB`. @bernd-ulmann

bernd-ulmann, 2020-08-19:
The programming card has been updated. :-)

MJoergen, 2020-08-19:
> @MJoergen Can you have a look at the first two checkboxes in the list above? I would then take the others.

Done.

sy2002, 2020-08-21:
Note to self I need to update the compiler documentation in doc/vbcc.txt, doc/constaints.txt and doc/best-practices.md to cover the ISR handling and `-speed`

```
__interrupt __rbank void irq(void)
{
...
}
```

sy2002, 2020-09-03:
Update the qnice-fpga.com website: new version number, mention hardware platforms, mention Vivado, add Michael to the contributors

sy2002, 2020-09-07:
Everything done. V1.6 is released.

closed 2020-09-07 by sy2002

MJoergen, 2020-09-07:
Impressive! Turns out V1.6 is a really big release. Congrats to all of us!