sy2002/QNICE-FPGA #65: Update VASM (and maybe VBCC's code emitter) to ISA 1.6

sy2002, 2020-08-15:
The C Compiler Toolchain is working right now with V1.6. The following task / issue is currently meant to be a V1.7 task, as it is "just" an optimization.

The [QNICE ISA 1.6](https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/intro/qnice_intro.pdf) changes the semantics of opcode `E` formerly known as `HALT` into a Control instruction that as up to five sub-commands, see pages 9 and 11 in the [PDF](https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/intro/qnice_intro.pdf):
```
HALT
RTI
INT
INCRB
DECRB
```
Our C toolchain is currently not aware of this: The assembler VASM does not recognize these mnemonics and VBCC's code emitter (if he needs to be aware of all this, which I do not know right now) is also not aware.

Still, it works fine, because `INCRB` and `DECRB` are modeled in `dist_kit/qnice-conv.vasm` as assembler macros for the old way of doing it using `ADD` and `SUB` instructions in conjunction with the `SR`. And since this old way of switching banks by doing e.g. an `ADD 0x0100, R14` (R14 = SR) still works, we are still compatible here.

But this is not optimal, so the following TODOs need to be done either by one of us or by Volker Barthelmann. I will ask Volker, if he wants to do it and can spare the time. In this case, we might be able to do it in V1.6 - otherwise this is a V1.7 task.

- [ ] Update VASM to recognize the new opcode `E` behaviour and output the right code
- [ ] Check how the C toolchain does register bank-switching and change it to the new `INCRB` / `DECRB`
- [ ] Recompile the QNICE monitor lib and the standard C library
- [ ] Recompile /qbin
- [ ] Perform tests

sy2002, 2020-08-20:
@MJoergen I made a new branch called `dev-vbcc-vasm-fix`: Would be cool, if you could write some VASM test program (and put it to `c/test_programs`) which tests if the VASM is now supporting all this and then work through the above mentioned check-list.   V1.6 would be a much "well rounded package" if it had the updated VASM and VBCC in it, because in V1.6 we introduced all these new stuff and would be strange if only on V1.7 the full toolchain would support it. Also VERSIONS.txt would need an update.

sy2002, 2020-08-20:
P.S. You need to run `tools/make-toolchain.sh` again to make sure you have the newest VASM and VBCC. (I already recompiled the standard C and monitor lib and checked it into the branch)

MJoergen, 2020-08-21:
I've verified the programs in `c/test_programs` with the new VBCC. They work, except for the Issue #83, which was a problem in the old VBCC too, so it's not a new bug introduced.

I've verified by examining the generated assembly code (using the `-k` switch) that function calls now use `INCRB / DECRB` instead of the old `ADD / SUB 0x0100, R14`.

I've written a small test program `vasm_test_isa_16.asm` that verified the new instructions added to the ISA V1.6

I conclude that the updated VBCC is working as expected with fewer bugs than before. Only two known bugs as of today: Issues #76 and #83 .

MJoergen, 2020-08-21:
I've now verified the programs in the `qbin` folder running on the hardware.

Closing this issue, as it is now ready to be merged into the `develop` branch.

closed 2020-08-21 by MJoergen