sy2002/QNICE-FPGA #66: Emulator: Convenient Debugger

sy2002, 2020-08-17:
As a result of our video conference last Saturday: For solving everything related to future ISA changes in V1.7 and later (project "clean ISA", issues #57 and #55 etc.) and the related potentially large debugging sessions of existing QNICE software, we need a convenient debugger. I am telling this from experience: It took me > 2hrs to find the `RBRA XYZ, V` problem in Q-TRIS using today's debugger. With the debugger I am describing here, it would have taken < 5min.

We need something like this:

[image]

* Source code view incl. single-step debugging directly inside the source
* Be able to load `.lis` files and single step while seing all the valuable comments of the original source
* Register view (incl. way to browse the banks forward/backwards)
* Special convenient status register view
* Stack view
* Multi purpose view for examining memory locations and for conveniently "watching" variables as in Borland's Turbo Debugger
* Arbitrary amount of breakpoints
* Conditional breakpoints (like "break here when R8=100")

I know that Bernd "truly loves" to code anything related to user interfaces that are not compatible with line printers :-) 😈 

Therefore I fear, that Bernd would never do this here. @bernd-ulmann Your thoughts? Am I knowing you well enough already or did I get this one wrong? 😄 If I got it wrong @bernd-ulmann then please assign this one to you.

We might use Curses / NCurses or an alternative of it (https://en.wikipedia.org/wiki/Ncurses) for a convenient text based user interface.

And just one more thing: We might even need a debugger on hardware, just in case we cannot reproduce every problem during fixing issues #55 and #57 on the emulator. But this is not in scope of this issue (we might create another issue about hardware debugging later).

In principle, I think I would find it interessting to write such a debugger as part of the emulator. But I do forsee a resource problem on my side beginning mid September. But well, we do not have a deadline for V1.7, so it might be that V1.7 takes a little while (as in "a year so so ;-)") and then this would not be a problem that I am doing it. But I do not want to be a bottleneck here, so: 

@MJoergen Just in case you are interessted in picking this one, please assign it to you.

Otherwise, after I heard both of your feedback and depending how your feedback is, I would assign it to me.

MJoergen, 2020-08-17:
I agree, we need something like this. I tried debugging the latest `cpu_test.asm` on hardware, and that was quite tedious.

Regarding `ncurses` I would love to have support for that. It would make porting existing C-programs to QNICE almost trivial in many cases. I do think this is the best long-term solution, i.e. to implement a `ncurses` library.

Regarding debugging in hardware, I've in previous projects used an VGA overlay, i.e. a virtual layer on top of the current VGA screen that displays all relevant state information of the processor. This would all be implemented in pure VHDL code, and could be enabled/disabled by one of the switches. It would be "simple" to e.g. disassemble ten lines around the current PC, to implement single-step, to set breakpoints, etc.

Something like that can be done in various stages, where the first stage is just to display the current value of R0-R15, the second stage is to disassemble the current instruction, etc. Controlling the execution could be done using the push-buttons.

We still have a way to go before V1.6 is finished, and then we have to decide on, what we would like to include in V1.7. Only then can we give a very rough estimate of when V1.7 might be finished.

sy2002, 2020-08-17:
@MJoergen Just to avoid misunderstandings: This issue here is about having a better debugger in the **emulator** and using `ncurses` in the emulator. Not yet on the hardware :-)

But your ideas with the VGA overlay are so cool - I opened a new issue "Debugging on Hardware" and copied your ideas there: See issue #67 

What are your thoughts on doing the software based debugger? Would you like to do it? Otherwise, if Bernd does not want to do it - I would... ...somewhere in V1.7 as a prereqisite of the "clean up ISA" project :-)

bernd-ulmann, 2020-08-17:
@MJoergen @sy2002 :

You are both right - such a debugger would be really nice to have (although I found the nasty pointer error bug with the timer simulation by staring at a long hex dump for a long time ;-) ). 

The problem is that I have quite little spare time on my hands and I am by far no expert when it comes to ncurses etc. 

Could we, maybe, somehow attract someone to help on the software-side of the QNICE-project? Isn't there a way to recruit people via git? (I seem to remember that there was something like this...)

What do you think?

sy2002, 2020-08-17:
Bernd, no problem :-)  I assumed that I would un-assign you from this task :-)   Just did it.   So this one will be either done by Michael or myself.