sy2002/QNICE-FPGA #68: SHL/SHR and N/Z bits

bernd-ulmann, 2020-08-17:
Hi Michael - 
Mirko and I just had a discussion regarding the shift instructions. It was implemented so that only the X and C condition flags were used/written while NZV were untouched.

 I now thought about classic multiplication/exponentiation algorithms where one typically shifts some value to the left/right until the register contains zero. This would require SHL and SHR to change N and Z after each instruction. 

 We discussed this back and forth and Mirko is not opposed to each of the two behaviour variants while I would prefer SHL/SHR changing N and Z, so I would like to hear your opinion on that. I am aware that this would (again - I am *really* sorry about this!) affect the cpu_test.asm but I think it might be worth the effort as it would save a CMP 0x000, ... within a shift-controlled loop.

 What do you think?

 Have a great evening, sincerely, Bernd. :-)

MJoergen, 2020-08-18:
Hmmm, I hadn't thought about that.

I agree with you @bernd-ulmann that setting the N and Z flags after a SHL/SHR would be beneficial. So I think we should go forward with this, and I'll change the `cpu_test.asm` accordingly.

BUT, will some existing programs break after this change ?

bernd-ulmann, 2020-08-18:
I am really happy that you, too, like that idea. I will change the emulator and see if anything else breaks (although I doubt it :-) ).

sy2002, 2020-08-18:
@MJoergen @bernd-ulmann Don't worry: The hardware in past always did set N and Z after SHL/SHR. So I do not think, that this change will break anything: I only "fixed" this behaviour in hardware yesterday evening. So rolling this back, will be fine.

@MJoergen I assign this one to you and as soon as you have the new cpu_test.asm please assign it back to me and I will revert the hardware to the old behaviour.

And then we can close this issue and finally have an agreed upon ISA for 1.6 😄

MJoergen, 2020-08-18:
`cpu_test.asm` updated. Reassigned to @sy2002 .

sy2002, 2020-08-18:
The hardware CPU implements the ISA changes described in this issue. The newest `cpu_test.am` is passing. I also did "The Smoke Test", which is also passing.

@bernd-ulmann Reassigned to you, as the emulator is not yet passing `cpu_test.asm`: Please Close and comment this issue as soon as you fixed that.

bernd-ulmann, 2020-08-19:
The emulator now, too, passes the CPU test. :-)

closed 2020-08-19 by bernd-ulmann