sy2002/QNICE-FPGA #70: Possibly misleading comments and table structure in cpu_test.asm

sy2002, 2020-08-18:
In [cpu_test.asm line 1270 onwards](https://github.com/sy2002/QNICE-FPGA/blob/develop/test_programs/cpu_test.asm#L1270) there is this code
```
; ---------------------------------------------------------------------------
; Test the SUB instruction

L_SUB_00        MOVE    STIM_SUB, R8
L_SUB_01        MOVE    @R8, R1                 ; First operand
                RBRA    L_SUB_02, Z             ; End of test
                ADD     0x0001, R8
                MOVE    @R8, R0                 ; Second operand
                ADD     0x0001, R8
                MOVE    @R8, R2                 ; Carry input
                [... more code ...]
                SUB     R0, R1
```

`SUB R0, R1` means that `R0` is the first operand in contrast to the comments above, and `R1` is the second. This alone would not be that confusing - you could just change the comments. But in fact the table that follows the test is not intuitive any more:

```
STIM_SUB        .DW     0x5678, 0x4321, ST______, 0x1357, ST______
                .DW     0x5678, 0x5678, ST____C_, 0x0000, ST___Z__
                .DW     0x5678, 0x5679, ST______, 0xFFFF, ST_VN_CX
```

Looking at the table, one would expect that you would do

`<pseudocode> SUB 0x5678, 0x4321`

but what you actually do is

`<pseudocode> SUB  0x4321, 0x5678`

Currently, everything works fine - and even the hardware passed cpu_test.asm - so there is no urgent need to change this slightly misleading (unintuitive) code.

So please decide if you go for a WONTFIX or for a fix in V1.6 or any time later.

P.S. If you decide for a fix, then the next question is: Are there more places like this in cpu_test.asm?

MJoergen, 2020-08-18:
Thank you for pointing this out. I actually struggled with this test. The reason is that the ISA (in my opinion) is slightly confusing, in that the instruction

```
SUB R0, R1
```

actually calculates R1 - R0. So what is now "first" and "second"? It's not a problem with ADD, because there the ordering doesn't matter.

I decided to let the values in the table be swapped, so that the "destination" value comes first, and the "source" value comes second. But I agree that it is confusing calling them first and second.

Perhaps I should use the words "minuend" and "subtrahend" in the comments. These words have precise meanings, but are perhaps not well known. Or should I just use the words "destination" and "source" in the comments ? What do you think ?

bernd-ulmann, 2020-08-18:
I like the minuend/subtrahend notation! :-)

MJoergen, 2020-08-18:
Fixed and closed.

closed 2020-08-18 by MJoergen