sy2002/QNICE-FPGA #72: Choose an Open Source license and add a LICENSE.md

sy2002, 2020-08-18:
Currently, we say nothing out the Open Source license. We kind "float around in open space".

If OK for you gentlemen, I'd like to put this whole project under the https://creativecommons.org/licenses/by/4.0/ where not otherwise stated (some third party parts use other licenses for their work).

What I particularly like about Creative Commons is that they have this shortcut website that they call "human readable" and that **attribution of the authors** is part of the license. And of course that all liability for anything is taken away from us.

https://creativecommons.org/licenses/by/4.0/

And then there is the long legalese part: 

https://creativecommons.org/licenses/by/4.0/legalcode

That means our LICENSE.md could just have the human readable part and link to both CC parts, e.g. like I did it at TeleBall, [have a look at it here](https://github.com/sy2002/TeleBall/blob/master/LICENSE.MD).

=====================
SAMPLE LICENSE.md THAT WE COULD USE:

The QNICE ISA, assembler, emulator and FPGA implementation as well as everything in this GitHub repository that is not licensed otherwise (as stated in file headers) is licensed according to the **CREATIVE COMMONS 4.0 INTERNATIONAL** license: https://creativecommons.org/licenses/by/4.0/

In short human readable form while not replacing the original license text in https://creativecommons.org/licenses/by/4.0/legalcode, this means:

### You are free to

* **Share** — copy and redistribute the material in any medium or format
* **Adapt** — remix, transform, and build upon the material for any purpose, even commercially.  

As long as you give appropriate credit to the `QNICE Development Team` and link to `https://qnice-fpga.com/`, provide a link to the license `https://github.com/sy2002/QNICE-FPGA/blob/master/LICENSE.md`, and indicate if changes were made. You may do so in any reasonable manner, but not in any way that suggests the licensor endorses you or your use.

bernd-ulmann, 2020-08-18:
That sounds very reasonable to me! :-)

sy2002, 2020-08-18:
Oops sorry gentlemen, I just found, that creative commons is not that good for software:
https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

Back to square one :-)

sy2002, 2020-08-18:
OK - it looks like the 3-Clause BSD License is a good fit:
https://opensource.org/licenses/BSD-3-Clause

## Here is my proposal for us

The QNICE ISA, assembler, emulator, documentation and FPGA implementation as well as everything in this GitHub repository that is not licensed otherwise (as stated in file headers) is licensed according to the BSD-3-Clause license as follows:

Copyright 2006-2020 The QNICE Development Team

Redistribution and use in source and binary forms, with or without modification, are permitted provided that the following conditions are met:

1. Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer.

2. Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the following disclaimer in the documentation and/or other materials provided with the distribution.

3. Neither the name of the copyright holder nor the names of its contributors may be used to endorse or promote products derived from this software without specific prior written permission.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
```

sy2002, 2020-08-19:
@MJoergen and @bernd-ulmann Please check if you are OK with this, then I would do the LICENSE.md and close this issue.

bernd-ulmann, 2020-08-19:
That sounds like a good license for our project! :-)

MJoergen, 2020-08-19:
I agree. Let's use this license.

sy2002, 2020-08-19:
OK, then so be it: DONE.

closed 2020-08-19 by sy2002