sy2002/QNICE-FPGA #74: Documentation enhancement

MJoergen, 2020-08-19:
I would very much like the `qnice_intro` document to mention, which instructions modify which flags. So far, the CMP instruction has been described, and there is a general statement (on slide 5) saying that the flags are updated by the "last result" or "last operation".

In particular I find it important that the MOVE instruction does modify N, Z, and X flags.

I'm thinking about a table something like this (taken from cpu_test.asm):

```
; Instruction | Flags affected
;             | V | N | Z | C | X |
; MOVE        | . | * | * | . | * |
; SWAP        | . | * | * | . | * |
; NOT         | . | * | * | . | * |
; AND/OR/XOR  | . | * | * | . | * |
; ADD/SUB     | * | * | * | * | * |
; SHL         | . | . | . | * | . |
; SHR         | . | . | . | . | * |
; CMP         | * | * | * | . | . |
; BRA/SUB     | . | . | . | . | . |
```

Ideally, it would be nice for V1.6, but I guess this is not a top priority.

bernd-ulmann, 2020-08-19:
Please have a look at the new qnice_intro.pdf. I hope I got the table right. :-) (I am a bit in hurry... Sorry!)

sy2002, 2020-08-19:
Bernd was super quick :-) Can be part of V1.6. Michael, please check, if this is what you wanted and close this here or assign back to Bernd.

MJoergen, 2020-08-19:
Verified and closed.

closed 2020-08-19 by MJoergen