sy2002/QNICE-FPGA #75: C compiler bug: Correct output only when at least "-O1" is on

sy2002, 2020-08-19:
This simple test program is supposed to output `4`:

```
#include <stdio.h>
  
char grid[16];

int main()
{
   int sq = 9;
   int dir = 2;

   grid[sq] = 0;

   grid[sq] += 1 << (dir);
   printf("%d\n", grid[sq]);

   return 0;
}
```

When compiling it using `qvc test.c` and running it, it outputs `8`: WRONG.

But when compiling it using at least optimization level 1 `qvc test.c -O1` or higher (`-O2` or `O3`), then it works and it outputs `4`: CORRECT

MJoergen, 2020-08-19:
Interestingly, usually compiler optimizations are the source of bugs, not the other way around :-)

sy2002, 2020-08-20:
@MJoergen Please retest with Volker's newest compiler in branch `dev-vbcc-vasm-fix`

sy2002, 2020-08-20:
@MJoergen In general, please also retest the sierpinski and wolfram demos, because the code emitter changes were subtle but influencial.

MJoergen, 2020-08-20:
Verified the original issue.

sy2002, 2020-08-20:
Verified means: It still fails? Would you like to approach Volker and send him the source code of the test program via email? As far as I remember, Volker is not active on GitHub.

MJoergen, 2020-08-20:
Sorry. I was not clear. I meant that the latest branch fixes this issue, so it no longer fails.

In a comment above you mention the other demos, so they still need to be retested.

sy2002, 2020-08-20:
Cool 👍   Feel free to close this one as soon as you feel comfortable with its stability.

MJoergen, 2020-08-21:
I've checked sierpinski and wolfram.c. They worked as expected. I found a problem in the arith.c program, so I made a separate issue for that.

Closing this issue now.

closed 2020-08-21 by MJoergen