sy2002/QNICE-FPGA #80: Scope for 1.7

sy2002, 2020-08-20:
Gentlemen, we might want to find a videochat (GoToMeeting) time slot in late September or early October to have a rough idea about what will be part of 1.7, since I have the gut feeling that the list tagged "V1.7" might proove to be too large 😄

MJoergen, 2020-08-27:
Here is a list of the open issues currently marked V1.7:

- [ ] #7 Optimize ROM/RAM usage
- [ ] #10 HW Breakpoints plus debugging support in Monitor
- [ ] #11 Add qtransfer as standard functionality to the Monitor
- [ ] #13 BS/DEL when entering hex digits (and at other reasonable places)
- [ ] #14 STDIN switch during input
- [ ] #17 Add multicolour support in VGA: Hardware and Emu
- [ ] #19 UART inconsistencies
- [ ] #20 Make the native assembler independent from an external preprocessor
- [ ] #29 Feature Request : Add FIFO to keyboard I/O
- [ ] #34 Refactor code to remove critical warnings in Vivado
- [ ] #35 Feature Request: Add interrupt when VGA retraces specific scan line
- [ ] #41 Simplify Portability and Maintenance and unify common VHDL code
- [ ] #42 Simulation: Timer Interrupt Hardware: Test Daisy Chaining Edge Case
- [ ] #45 Timer Interrupt Hardware: Check Edge Cases and Stability
- [ ] #55 CPU bus functional test
- [ ] #56 Document architectural basics
- [ ] #57 Revisit the ISA, decide finally and debug/fix the existing software
- [ ] #58 Webassembly target of Emulator: Make the timer interrupt work there
- [ ] #60 Calculate all the hardcoded bit-widths automatically
- [ ] #62 Nasty edge case with timer and memory load via UART
- [ ] #63 Making the timer emulation more robust
- [ ] #66 Emulator: Convenient Debugger
- [ ] #67 Hardware: Convenient debugging
- [ ] #77 VBCC: Test the ISR feature and write a timer interrupt/ISR test for VBCC
- [ ] #79 Add Yamaha Sound Chip
- [ ] #81 CPU speed-up "Fastpath": Direct register to register operation
- [ ] #86 Making qtransfer should be done automatically
- [ ] #87 Maze: Show your gamers some more love :-)
- [ ] #88 Assembler parses comments
- [ ] #90 MEGA65: HyperRAM: 16-bit write/read not stable
- [ ] #91 MEGA65: HyperRAM: Very first control register read needs two reads

MJoergen, 2020-08-27:
Here are issues that have already begun, or perhaps even almost complete:
- #17: Already started. Will be part of V1.7
- #81: Almost done. Needs documentation. Will be part of V1.7
- #34: Is there anything else that needs to be done here?
- #41: Not sure what is left to be done here either.