sy2002/QNICE-FPGA #82: emulator crashes on exit

charlesUnixPro, 2020-08-21:
Following the instruction in emulator/README.txt, at the step:

>  Press `CTRL+D` or enter `exit` to end the emulator.

we seeL

> Q> Segmentation fault (core dumped)

The issue is sd_detach is being called when image is null.

Fix:

    void sd_detach()
    {
    -  fclose(image);
    +  if (image)
    +    fclose(image);
      image = 0;
      memset(sd_data, 0, SD_SECTOR_SIZE);
    }

sy2002, 2020-08-21:
@charlesUnixPro Thank you, Charles for reporting this one.

You are right, and staring at this code I am indeed wondering, why it never crashed on our machines 😄 

We are about to release V1.6 (branch `develop`) soon, and we will fix the issue there. (@bernd-ulmann You might want to look after that - if you are time constrained, just assign it to me)

bernd-ulmann, 2020-08-21:
Done. :-)

closed 2020-08-21 by bernd-ulmann