sy2002/QNICE-FPGA #83: '<<' in arith.c gives wrong result

MJoergen, 2020-08-21:
The following is an excerpt from the output of the arith.c program, when given the values 5 and 7:
```
signed arithmetics
5 << 7 = -8387968
unsigned arithmetics
5 << 7 = 4286579328
```

In both the signed and unsigned cases the expected value is 5*2^7, i.e. `640`.

sy2002, 2020-08-21:
I guess that this one is related to issue #76, because in an email, Volker wrote:

> Mirko: Michael detected another pretty cool compiler bug:

> Volker: That one was tricky. I do not have the board set up at the moment, so I
used the emulator. It seems that the carry flag is not set by most
instructions in the emulator. AFAIR the FPGA did reset the carry flag
for most instructions and vbcc uses or r13,r13 to clear the carry flag.
This does not work with the emulator.

>If I change the backend to use add 0,r13 and rebuild the library, the
test apparently works in the emulator.

>Does the test also fail on the FPGA? Which is the intended behaviour?

That means, @MJoergen and @bernd-ulmann our changes to the processor flag behaviour lead to a "destruction" of a lot of things in the VCCC backend.

We need to fix that before V1.6 (which means that I will take my FPGA with me in the vacation! :-) ), and this means we probably will not release today.

sy2002, 2020-08-21:
Here is my email to Volker:

Thank you for your detailed answer and thank you SO MUCH for finding this issue with the Carry flag!

I was a bit desparate in the last two days, because I did remember VBCC to be rock solid.

And now, as Michael starts using it, he is finding absurd cases, that are not working any more, for example look at this!

https://github.com/sy2002/QNICE-FPGA/issues/83

And I was like: WHAT? Hell! That worked in past like a charm!

And then you came up with your email and literally made my day. :-)

You wrote:

==== VOLKER ====
That one was tricky. I do not have the board set up at the moment, so I
used the emulator. It seems that the carry flag is not set by most
instructions in the emulator. AFAIR the FPGA did reset the carry flag
for most instructions and vbcc uses or r13,r13 to clear the carry flag.
This does not work with the emulator.

If I change the backend to use add 0,r13 and rebuild the library, the
test apparently works in the emulator.

Does the test also fail on the FPGA? Which is the intended behaviour?
==== VOLKER ====

And then it made "click": Yes, we did change the behaviour of all instructions, when it comes to flags, to be more consistent. And we fixed a bug on the hardware: The hardware was "forgetful" and deleted the carry flag for example when you did a MOVE.

The V1.6 ISA breaks all code pre-V1.6 that has to do with flags.

Here is the new behaviour when it comes to flags:

wget https://raw.githubusercontent.com/sy2002/QNICE-FPGA/develop/doc/intro/qnice_intro.pdf

Go to page 12/13.

We now also have test_programs/cpu_test.asm, which is a very detailed CPU test that ensures that the emulator and the hardware are 100% behaviorally identical. That means, if you test in the emulator, you can be very confident that it works in the CPU.

So, here is the question for you Volker:

Now that you now about this flag behaviour: Are there more spots that come to your mind in the context of the absurd failing of nearly all mathematical functionality currently in the compiler, such as

https://github.com/sy2002/QNICE-FPGA/issues/83
and
https://github.com/sy2002/QNICE-FPGA/issues/76
?

Could you please deliver a new machine.c (or others)?

@Bernd: Is ADD 0, R13 the best way to clear carry? Is maybe instead AND 0xFFFB the better way to do it?

*phew* I am sooooo happy that there is finally a light at the end of this "VBCC instability tunnel" ;-)

sy2002, 2020-08-21:
We do everything related to that on branch `dev-vbcc-vasm-fix`.

bernd-ulmann, 2020-08-21:
I, personally, would clear the Carry flag with an AND-mask:

AND 0xFFFB, SR

:-)

sy2002, 2020-08-21:
Typo in https://github.com/sy2002/QNICE-FPGA/issues/83#issuecomment-678108835: AND 0xFFFB, R14 (because R13 is the stack pointer)