sy2002/QNICE-FPGA #96: Add a stereo SID (and maybe a 6510 co-processor)

sy2002, 2020-08-29:
### Add a stereo SID (and maybe a 6510 co-processor)

The goal is to play SID tunes. It's been a while, but as far as I remember, SID tunes contain 6510 code, so we either need to emulate that in assembler of have a 6510 coprocessor.

There are great sounding SID VHDL implementations available (for example the one used in MEGA65). So this is more about integrating an already existing open source SID, than to do a new one. As for the 6510, I would be very interessted to write one by myself instead of using an existing one - but maybe due to time reasons, this might have to wait :-)

MJoergen, 2020-11-30:
Here is another SID implementation (in verilog): https://github.com/thomask77/verilog-sid-mos6581

MJoergen, 2020-11-30:
There is already a working 6502 processor here: https://github.com/MJoergen/65c02 (made by me!). That implementation is not quite cycle exact, but could be made so, if the need is there.

sy2002, 2020-11-30:
Just in case you would like to tackle this issue: Feel free :-)