sy2002/QNICE-FPGA #100: Update to new compiler version, recompile all libraries and test

sy2002, 2020-08-31:
### Update and recompile the compiler

1. Merge the branch `develop` into `dev-vga-colour`. Reason: The newest commits of `develop` are storing our specific improvements and fixes of the compiler backend (including the bugfixes for clearing the X and C flags that lead to the nasty arithmetic errors) in a separate folder: `c/qnice/compiler-backend/`. This makes sure, that we are not loosing our newest compiler backend code, when applying a patch from Volker's main repository. Also, the `develop` branch contains updated make files that make use of these changes.

2. Get the newest snapshot of VBCC and copy it to `c/vbcc`

3. Run **`tools/make-toolchain.sh`**. This makes sure, that `monitor/compile_and_distribute.sh` is automatically being run first, which is important for having the newest `sysdef.h` in `compiler-backend`. From there, the new compiler backend is automatically copied into the VBCC folder.

###  Recompile the libraries

1. Go to the `c` folder and enter `source setenv.source`

2. First run `make-qnice-monitor-lib.sh`, because the standard c library depends on it

3. Then make sure that you have copied the source code into `vclib`

4. Then run `make-vclib.sh`

### Recompile /qbin and test

1. Run `/qbin/make.sh`

2. Test what can be tested in the emulator

3. Test on hardware

4. At least also test `c/test_programs/arith.c` and your bug-test-c files to make sure, all our bugfixes made it to the final result

### Check in everything 

Check in everything, commit and push to GitHub

sy2002, 2020-09-29:
There is an updated version of the documentation of how to update the compiler: Use this one:

https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/vbcc.md#updating-to-newer-compiler-versions

sy2002, 2020-10-03:
I updated to the newest VBCC: While doing so, I discovered, that my compile script had a bug and that the documentation was also not precise enough when it comes to keeping the old `config` and `targets` folder. Fixed that, see: https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/vbcc.md#update-and-recompile-the-compiler

Then I did a very superficial smoke test: Only on the emulator, not on hardware.

**Assigning to @MJoergen Please test on hardware, before closing this issue.**

Here are the results of my smoke test (I tested the /qbin apps that are made with C):

|App|Result|Comment
|----|------|----------
|adventure.out| ✅ 
|demo_atom.out| ✅ 
|demo_sprite_balls.out| ✅ 
|demo_tree.out| :x: | Looks weird, see issue #151 
|hdmi_de.out| :x: | Not supposed to work in current emulator version. Possibly deprecated due to new VGA
|maze2d.out| :x: | Stops after maze has been generated, see issue #152
|shell.out| ✅ 
|sierpinski.out| :x:
|ttt2.out| ✅ 
|vga_calibration.out| ✅ 
|wolfram.out| ✅ 
|world.out| ✅

sy2002, 2020-10-11:
After @MJoergen added the VBCC bugfix, I recompiled the monitor lib, the compiler's standard C lib and /qbin. Still left as TODO is at least a smoke test like https://github.com/sy2002/QNICE-FPGA/issues/100#issuecomment-703107614 on emu and hardware needs to happen.

MJoergen, 2020-10-13:
Following is tested on hardware:
* maze2d
* demo_tree
* demo_sprite_balls
* world
* demo_atom
* mandel

hdmi_de is deprecated due to new VGA

Closing this issue now.

closed 2020-10-13 by MJoergen