sy2002/QNICE-FPGA #104: VGA scanline interrupt: Behave like a good Daisy-chained citizen

sy2002, 2020-09-04:
**Currently, the way VGA is implemented, it ignores the mechanisms of our Daisy-chained architecture. So the TODO of this issue is to implement it. This issue blocks issue #17**

MJoergen, 2020-09-07:
Thank you for the detailed description.

As mentioned in Issue #108, the current VGA interrupt mechanism is completely broken (it corrupts the data bus), and I'm in the process of re-implementing it according to the above description. However, I'm running into some problems, so I would like to have some things clarified related to when the ISR address is being presented to the CPU.

In the following I'm referring to this diagram:

[image]

Here is my understanding:
* The device pulls `INT_N` low when it wants to signal an interrupt.
* Some time later the CPU pulls `GRANT_N` low and then waits for the ISR address from the device.
* Some time later the device pushes the ISR address onto the data bus and simultaneously releases `INT_N`.
* Some time later the CPU reads the ISR address, releases `GRANT_N`, and then starts fetching the next instruction.
* IMMEDIATELY (i.e. combinatorially), the ISR device must release the data bus, so as not to interfere with the next instruction fetch.

So it appears there is a requirement that the `GRANT_N` signal be passed combinatorially through the entire daisy chain. This requirement should be made more explicit in the description of the daisy chain process. This requirement is seemingly not obeyed by the TImer device in `timer.vhd`.

I notice that when the VGA module is connected as the last device in the daisy chain, the `GRANT_N` input to the VGA module is delayed some clock cycles compared to `GRANT_N` coming from the CPU, and therefore when the CPU releases `GRANT_N`, the VGA module is still pushing the ISR address onto the data bus, which leads to corruption.

I would presume the same problem exists with the Timer 1 module since it too receives a (slightly less) delayed version of `GRANT_N`. I have not yet tried to confirm this experimentally, but it could be as easy as placing a Timer 1 ISR around address `0xE500` like in Issue #108.

MJoergen, 2020-09-07:
And a more general comment about the interrupt daisy chaining architecture:

Currently we have two timers and a VGA module. Soon we will have an Ethernet module too. All these devices are required to adhere to a very strict handshaking protocol whose implementation is non-trivial. So as not to have to spend too much time debugging hardware interrupt implementation, I have attempted to make a general daisy chaining module [vhdl/daisy_chain.vhd](../blob/dev-vga-int/vhdl/daisy_chain.vhd) in a new branch `dev-vga-int`.

The idea is that this module is responsible for all the intricacies of the daisy chaining protocol, whereas the individual devices (Timer, VGA, Ethernet, etc.) may remain agnostic about the overall interrupt architecture of the system. You can see an example of how this is connected in the file [vhdl/vga_multicolour.vhd](../blob/dev-vga-int/vhdl/vga_multicolour.vhd).

In other words, future devices (Ethernet, etc) who want to issue interrupts, should instantiate this daisy chaining module, and then everything work out-of-the-box.

sy2002, 2020-09-07:
Excellent thoughts!

**I particularly love your idea about having a general daisy chain module that everybody can and should use.**

So here is my proposed the roadmap about this issue:

- [ ] You and I discuss all aspects of https://github.com/sy2002/QNICE-FPGA/issues/104#issuecomment-688019689 and https://github.com/sy2002/QNICE-FPGA/issues/104#issuecomment-688023695 and you walk me through your code. Four eyes priciple. Preferably today evening 21:00 in our online meeting

- [ ] **V1.6**: Both of us in a screensharing session (also preferably today) update the `doc/int-device.md`, so that we have a 100% reliable "law" documentation in `develop`.

- [ ] **V1.7**: I am updating the timer interrupt modules using your `daisy_chain.vhd` in `dev-vga-colour` and in the V1.7 version of `doc/int-device.md` we should also mention the module `daisy_chain.vhd`, describe its usage and only provide the documentation of how to do an own version for future reference. Some sentence like "Implementing the correct daisy chain protocol is not trivial and can lead to non obvious hard to find bugs. This is why we suggest that you use our rock solid and proven to work `daisy_chain.vhd` instead of creating your own version."

MJoergen, 2020-09-08:
Fixed in the branch dev-vga-int.

sy2002, 2020-09-09:
Done with commit e9d4f3deeafca2e4766806e924437044795003bf

I did some extended smoke tests on hardware, here is what works:

✅ **The Smoke Test** (including loading Q-TRIS via SD/FAT32 while the timer interrupt is running) 
✅ **fancy** stand alone, mode `F` and then mode `B`
✅ **fancy** in parallel with **Q-TRIS** (but due to issue #109, Q-TRIS was already in memory)
✅ **large file load: qbin/adventure.out** from SD Card while `test_programs/timer_test` is running and playing it

Here is what does not work, but these are separate issues that have been there before the Daisy chain module was implemented in the timer that do not prevent this one from closing:

* issue #65 Nasty edge case with timer and memory load via UART
* issue #109: test_programs/fancy.asm does not work in parallel with SD-Card / FAT32 or with qtransfer

closed 2020-09-09 by sy2002