sy2002/QNICE-FPGA #105: Palette handling in Monitor

sy2002, 2020-09-05:
Goal of this issue:

1. The Monitor's look shall be as usual: "Full green".

2. But as soon as a program is being started using `C` / `R` or `F` / `R`, the Monitor will make sure, that the application can rely on having the **new nice palette** (which means: another green tone plus more colors). I plan to use the "bit 15 feature" to set the palette to the new default mode (**)

3. `SYSCALL`-ing `exit` will then revert back to the "Full green" before the `QMON>` prompt is being shown.

regarding (**):

@MJoergen in https://github.com/sy2002/QNICE-FPGA/issues/17#issuecomment-680224075 you wrote:

> Both the Font RAM and the Palette RAM can be reset to the default Font and default Palette by setting bit 15 of the corresponding address register. This bit will automatically clear when the reset is complete.

Looking at your code, I did not find this. It is also not mentioned/documented in `monitor/sysdef.asm`, so I guess, this is not implemented, yet?

MJoergen, 2020-09-06:
> Looking at your code, I did not find this. It is also not mentioned/documented in `monitor/sysdef.asm`, so I guess, this is not implemented, yet?

That is almost correct. Currently, the monitor ROM (in the file `vga_library.asm`) configures the default palette after reset. In other words, this is already done, but in software rather than hardware.

I suggest that the `exit` function calls the already existing function `VGA$INIT_PALETTE`. Would that solve this issue ?

sy2002, 2020-09-06:
@MJoergen Ah - OK, I was not aware, that you implemented the default palette via software. In the good old days (VGA and the likes) this was done in hardware: When you reset a VGA/EGA/CGA card - or an Amiga - or a C64, then the default palette was there - also when you decided not to use the operating system or basic ROM (in C64). As a "retro guy", I would prefer that.

But I can also live with software.

But then you need to "publish" your "operating system" function correctly:

1. You need to extract the function that is currently only a label within `VGA$INIT` and make an own real function out of it; for example as you suggested `VGA$INIT_PALETTE`. The existing `VGA$INIT` would of course then call `VGA$INIT_PALETTE`.

2. You then need to stick to the documentation conventions, example:
```
;
;***************************************************************************************
;* VGA$CHAR_AT_XY
;*
;* R8:  Contains character to be printed
;* R9:  X-coordinate (0 .. 79)
;* R10: Y-coordinate (0 .. 39)
;*
;* Output a single char at a given coordinate pair.
;***************************************************************************************
;
VGA$CHAR_AT_XY          INCRB
                        MOVE    VGA$CR_X, R0
                        MOVE    R8, @R0
... more code ...
```

3. When you stick to the conventions, then the **documentation is generated automatically**: We have a script called `monitor/create_documentation.pl` that collects all these functions and documents them in [doc/monitor/doc.pdf](https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/monitor/doc.pdf).

4. You need to go to `monitor/qmon.asm` to this place:
```
f32_fseek!      RBRA    FAT32$FILE_SEEK, 1
gets_s!         RBRA    IO$GETS_S, 1
gets_slf!       RBRA    IO$GETS_SLF, 1
vga_init!       RBRA    VGA$INIT, 1
<here is where you need to go, it is important that new functions always are added AT THE END of the list>
```
And then have something like (the exclamation mark is important):
```
vga_init_pal!      RBRA    VGA$INIT_PALETTE, 1
```
This makes the function `SYSCALL`-able by adding it to `monitor.def` and makes sure, that `./compile_and_distribute.sh` is putting it into all the different files for VASM, VBCC, etc.  That means from this moment on, the function can by used by `SYSCALL(vga_init_pal, 1)`. (At a later stage, when we separate user mode from kernel mode, SYSCALL will be more than a wrapper to `ABRA`.)

5. If you want to make this function available in C, then you need to extend the monitor library. A trivial function like `vga_init_pal` that does not need any input/output parameters to be processed so that C can digest it, can be very easily added to the header file of the lib, so not even a recompile of the lib is necessary:
Go to `c/qnice/monitor-lib/include/qmon.h` and go to here
```
#define qmon_puthex(x)              ((_qmon_fp)  QMON_EP_PUTHEX)(x)
#define qmon_str2upper(x)           ((_qmon_fp)  QMON_EP_STR2UPPER)(x)
#define qmon_exit                   ((_qmon_fp)  QMON_EP_EXIT)
<here is where you need to go>
```
and add a new define there, something like
```
#define qmon_vga_init_pal  ((_qmon_fp)  QMON_EP_INIT_PAL)
```
Note, that the entry point definitions ("EP" in `dist_kit/qmon-ep.h`) are generated automatically when you call `monitor/compile-and-distribute.sh`

6. After all that: Tell everybody to call `monitor/compile_and_distribute.sh` :-)

**I did assign this issue to you so that you can use this rather simple case to practice "how to make a new official operating system function that is available via SYSCALL and via C". After you have done and tested it, please assign this issue back to me.**

Furthermore, while writing this explanation I noticed: I should document all this in more detail ;-) So I created issue #110 and assigned it to me.

sy2002, 2020-09-07:
https://github.com/sy2002/QNICE-FPGA/issues/105#issuecomment-687870293 is obsolete now due to https://github.com/sy2002/QNICE-FPGA/issues/107#issuecomment-688511968

MJoergen, 2020-09-08:
I've now removed the Palette Initialization from the Monitor, because this is handled by the hardware.

@sy2002 : Is there anything else to be done in this issue? Currently, when pressing the Reset button, the hardware resets all the VGA configuration registers to default value (zero).

sy2002, 2020-09-08:
I have implemented now this specification: https://github.com/sy2002/QNICE-FPGA/issues/105#issue-694194700
So we can close this issue.

closed 2020-09-08 by sy2002