sy2002/QNICE-FPGA #106: Assembler bug: Non-existing mnemonics are accepted without an error

sy2002, 2020-09-05:
When you write something like this:

```
                    ; Disable hardware cursor
                    MOVE    VGA$STATE, R0
                    MURX    VGA$EN_HW_CURSOR, R1
                    NOT     R1, R1
```

Then the assembler will rightfully complain, because MURX is not really a mnemonic.

But when you do something like this:

```
                    ; Disable hardware cursor
                    MOVE    VGA$STATE, R0
                    MURX
                    NOT     R1, R1
```

or something like this

```
gone            DECRB                           ; restore R8..R11
                MOVE    R0, R8
                MOVE    R1, R9
                MOVE    R2, R10
                MOVE    R3, R11
                DECRB
                RTX
```

then neither `MURX` nor `RTX` is detected.

So as a summary: The assembler rightfully complains, if you have a non-existing mnemonic that is having parameters (such as `MURX R0, R0`, but it will not complain, if you have a non-existing mnemonic that has no parameters.

If you consider your fix as harmless and save, then please check-in into `develop`, then it would be part of V1.6.

Otherwise, please check-in into `dev-vga-colour`, which is our current V1.7 development branch.

MJoergen, 2020-09-06:
I think the problem is that the assembler interprets invalid mnemonics as a label. So in the examples mentioned, `MURX` and `RETX` are considered labels by the assembler. Presumably, you should be able to reference `MURX` and `RETX` as labels in the examples given.

So I don't see how to fix this, because currently a label is defined as anything that is not a valid mnemonic.

If we were to require all labels to be followed by a ":", then we could fix this issue, but that would require changing EVERY assembler file (albeit in a trivial way). I'm sure we could write a small script that adds all the necessary :'s automatically.

bernd-ulmann, 2020-09-06:
Michael is (of course :-) ) right, the assembler treats everything which it cannot recognise as a label. If this behaviour turns out to be a problem we could either introduce a ':' after a label or force labels to start in column 1 of the source. 

Without having thought too much about this, I, personally, would opt for the column-1 variant.

bernd-ulmann, 2020-09-06:
I just changed the assembler's behaviour so that mnemonics now must start in column 1 of a source file. This change now correctly recognises Mirko's examples as erroneous and still assembles the monitor etc. correctly. I have not tested this extensively (to say the least), so please have a look if I have broken something...

sy2002, 2020-09-07:
@bernd-ulmann For the future: When you want to have a quick check that goes beyond assembling the monitor, just go to `qbin` and run `make.sh`. This will assemble our whole "suite" of test programs.

I did that and indeed ran into a problem in `demos/tile_ed.asm` (this is an editor to develop font-based graphis that I used when creating q-tris back in the days). But this problem was "home made": I had one label accidentally start at column 2 instead of 1 (or 1 instead of 0). So: All good, I am closing this issue.

But @bernd-ulmann I ran into another problem that I would call a "convenience problem": Here is what happened:

```
assembling: tile_ed.asm
assemble: Line 885: No valid mnemonic/no label (does not start in column 1)
assemble: Line 1279: Unresolved label or equ >>END<<!
main: There were 2 errors during assembly! No files written!
An unrevoverable error occured!
```

Now, the problem is: my tile_ed.asm does not have a line 885 neither does it have a line 1279. This is of course happening due to the include files. But: Due to the fact that no `.lis` file is written in case of an error, as a programmer you need to start guessing: Where could the problem be. Would a solution maybe be just to have a `.lis` file also in case of error?

I am putting all of this into a new issue.

closed 2020-09-07 by sy2002