sy2002/QNICE-FPGA #107: VGA: A reset should reset the VGA to factory defaults

sy2002, 2020-09-06:
If one modifies the cursor (e.g. no blinking) or sets up a scanline ISR, then a system reset will not reset VGA back to factory default.

Example: Run `test_programs/vga_int.asm` and then press reset (instead of ending the program normally): The scroller will still be running.

This issue blocks issue #17.

MJoergen, 2020-09-06:
How much of this resetting should be done in hardware, and how much in software (in the function `VGA$INIT` in `vga_library.asm`)?

bernd-ulmann, 2020-09-06:
Just my two cents worth: I think initialization should be done in software mainly. When I think of Chris' TTL implementation of QNICE, this won't even initialize all the registers to sensible values as this would require quite a lot of microcode to accomplish.

I would ask every program to do all the initializations to IO devices which it needs. We can add this to the monitor so that it is run on a cold start and could be called as a system call as well, but I would not make it part of the exit-call.

sy2002, 2020-09-06:
Well, I was socialized in a world that you'd call today "retro home computers": C64 then PCs. And there it was like this: When you programmed some crap. A nasty bug and pressed the RESET button on your machine (PC had one built-in, C64 was an easy soldering excercise ;-)), then you were guaranteed that your machine works - also in cases where you decided to work without any operating system.

So, all rogue ISRs were terminated, the default font was there (even if you were about to program a computer game that trashed the font due to font graphics). The default palette was there. The cursor was blinking. 

**So my minimum requirement would be: Terminate all ISRs on a hardware reset by setting the ISR register to zero, so that a reset is somehow guaranteeing that regular programming errors in an ISR do not "brick" your computer. **

When it comes to the default palette - as written in https://github.com/sy2002/QNICE-FPGA/issues/105#issuecomment-687870293 I can live with initializing the palette in software (even though it does not feel as "retro home computer as it should ;-)").

But what about the font then? As we use font GFX, after pressing reset, the original font should be there again. Making the monitor ROM huge, just so that it is able to load the font from software - does not feel very good. The Monitor ROM should be as small as possible to leave as much core RAM as possible. Or requiring always an SD card to be present to load it from a file - also quite nasty. The default font should be in a font ROM (we did it like that during the last years).

sy2002, 2020-09-07:
Additional idea: We might put palette init and font init into the PORE ROM, which is executed upon each reset.

Advantage: Creates the retro feeling that a strive for (RESET=ready to go, even without an operating system) and does not require lenghty init routines in the hardware for font and palette and does not enlarge monitor rom. Sounds like the perfect solution...

MJoergen, 2020-09-07:
@sy2002 Sounds like a good idea with the PORE ROM. After all, that is exactly what it is for, if my understanding is correct.

sy2002, 2020-09-07:
As Michael and Mirko discussed in the online meeting, we are doing it like this:

* Have palette and font RAMs that are double the size, first half is treated as "ROM", second half is treated like RAM
* Have two new registers - something like "palette offset" and "font offset"
* Have four constants in `sysdef.asm` something like "USE_DEFAULT_PAL" and "USE_CUSTOM_PAL" and "USE_DEFAULT_FONT" and "USE_CUSTOM_FONT"
* Remove all palette setting code from Monitor
* The new registers should be set to "USE_DEFAULT_PAL" (which might be 0) and "USE_DEFAULT_FONT" (which might be also zero)

And: Set the ISR to zero on reset, too.

MJoergen, 2020-09-08:
When pressing the Reset button, the hardware automatically resets the entire VGA register map => Closed.

closed 2020-09-08 by MJoergen