sy2002/QNICE-FPGA #108: 0xE500 as start address leads to strange VGA ISR behavior

sy2002, 2020-09-06:
I assigned this one to MJoergen and sy2002 because it is unclear, if this is a CPU bug or a VGA bug or even something else.

Change line 11 in `test_programs/vga_int.asm` to

```
                    .ORG    0xE500
```

And then run `C` `R` `E500`: Screen will go black and even after a reset, there is no way to type anything anymore.

The obvious thing was: Is there a problem with the stack? But the stack pointer starts at `0xFEEB`, which is pretty far away from the last address of `vga_int.asm` which is - according to `vga_int.lis` `0xE687` and therefore still very far away from the stack.

@MJoergen Do you have any restrictions on the ISR address space in the VGA module? Or any other idea what the reason could be?

This issue blocks issue #17

MJoergen, 2020-09-06:
Good catch. I have no idea at the moment, but I'll fire up the ILA and have a look at it.

MJoergen, 2020-09-06:
This is a hardware bug in the VGA module. Here is what our beloved ILA shows:

[image]

As can be seen the VGA module outputs the ISR address E5FC on `VGA/data_o` twice. Once at cycle 520 correctly, and again at cycle 524 incorrectly. The second time is while the CPU is reading an instruction opcode, which then gets interpreted as HALT.

We can also see that the VGA module pulls `VGA/int_n_o` low a second time at cycle 522, before `VGA/grant_n_i` has gone high. This is breaking the handshaking protocol.

This issue will be fixed together with Issue #104.

MJoergen, 2020-09-08:
Fixed in the branch dev-vga-int.

closed 2020-09-08 by MJoergen