sy2002/QNICE-FPGA #109: test_programs/fancy_bg.asm does not work in parallel with SD-Card / FAT32

sy2002, 2020-09-06:
Reproduction:

1. Start `test_programs/fancy_bg.asm` 
2. Try to browse a directory or load Q-TRIS from `/qbin`

=> It will not work; the system will crash or behave strangely

An obvious reason might be, that the ISR of `fancy_bg.asm` is not leaving all registers untouched.
Other reasons might include hardware errors.

sy2002, 2020-09-06:
Might issue #108 be also the reason for this here?

sy2002, 2020-09-07:
Maybe also this discussion is related: https://github.com/sy2002/QNICE-FPGA/issues/104#issuecomment-688019689

sy2002, 2020-09-12:
Note to self/idea to be checked when we tackle this: after being pretty sure that it is not a rogue register: might it have something to do with SHR or SHL inside fancy changing X or C or inside other routines being disturbed by X or C?

MJoergen, 2020-09-12:
One idea for debugging is to start the `fancy.asm` ISR and then run the `cpu_test.asm` program multiple times in a row. The CPU test makes heavy use of the flags, and this should detect any hardware errors (I hope).

sy2002, 2020-09-14:
I did not test anything, yet, but here is another note to self/idea that might be the reason , adding @bernd-ulmann and @MJoergen to the discussion - you just need to read this here, no need to read the whole history of this issue:

* We have a great feature on our QNICE system architecture, which is register banks.

* In 90% of the cases, programmers might use the register banks instead of the stack, when entering / leaving a subroutine:
```
DRAW_WORKSPACE  INCRB
                [... a lot of more code ...]
                DECRB
                RET
```

* But hey, this is just one way to use this great architectural feature. You can also use register banks within routines to increase performance, because you use registers instead of variables, as we have so many of registers :-) 

* But our register banks could also used like this:
```
DRAW_WORKSPACE  INCRB
                [... fill the first bank ...]
                INCRB
                [... fill the second bank ...]
                INCRB
                [... fill the third bank ...]

                [... think about a lot of code here ...]

                DECRB
                DECRB
                [... now we need a register from the first bank for some calculation ...]
                [... so lets fetch it ...]
                [ (**) consider an interrupt happens here]
                INCRB
                INCRB

                [... Lets go on with our code ..]

                DECRB
                DECRB
                DECRB
                RET
```

* Now consider, that an interrupt happens at (**) and the ISR is doing an INCRB / DECRB to save the registers: BOOM. It will overwrite the second bank.

* I guess situations like this are the reasons, why a lot of processor architectures I know are able to clear some interrupt enable flag : We could put up some rules that say: If you are using INCRB INCRB INCRB situations, then prevent any interrupts until you are done with that.

* And/or we could set up a convention that stack and regbanks are not allowed to be used in interrupts, but this is also not very nice

Thoughts?

MJoergen, 2020-09-14:
I see your point. Currently, we can only access the *current* register bank, but in your example you want to access a *previous* register bank. We can't do that without explicitly modifying R14, and that leads to a potential race condition with an interrupt.

If instead values are stored on the stack (i.e. in main memory), then we can access those values by copying R13 (the SP) to a different register and doing some arithmetic to calculate the address of the stored data. The point being, that this operation does not *change* R13 and therefore interrupts can still happen transparently without corrupting data.

You mention a solution where interrupts are disabled while we temporarily rewind the register bank. That was my first thought too. However, the problem with the feature you mention is not only interrupts. It's also if the programmer wants to call a subroutine while the register bank is "rewinded". This will have the same destructive effect.

Another solution is to give the programmer the ability to access other register banks without *modifying* R14, e.g. by supplying an 8-bit offset to the current register bank. But I guess the current ISA does not support this, and I'm not sure whether we want that kind of support.

Yet another solution is to just not use "temporary rewinding of the register bank". This is probably my preferred solution (after thinking for less than 5 minutes).

sy2002, 2020-09-14:
> Yet another solution is to just not use "temporary rewinding of the register bank". This is probably my preferred solution (after thinking for less than 5 minutes).

I do not like this "solution".

Here is the reason: It demotes the whole register bank feature to just being usable in the context of INCRB/DECRB at the beginning of a sub-routine / function. That that is a bit too little as a use case for such a great and big architectural feat.

Your argument about the stack pointer holds.

That means the solutions circle around three thinigs - as far as I can oversee it:

1. We could set up the convention, that ISRs **must not** use register bank switching at all but use only the stack.
I guess this might be the easiest / short term "fix". The assembler programmer must not forget this (we need to add it as a best practice in our `doc/best-practices.md`). And: We must also make sure, that C programmers then always add the `__norbank` directive to their `__interrupt` directive - or, even better, besides mentioning this in `doc/best-practices.md` we might want to have a `#define __isr __interrupt __norbank` in our `sysdef.h` and tell everybody to use `__isr` instead of `__interrupt`, so that nobody is ever forgetting this.

2. We could have a an extension of the ISA, that let's us do similar thing with the register bank as we are able to do with the stack today.

3. We add the ability to temporarily halt all interrupts some `CLI` (Clear Interrupt Flag) `STI` (Set Interrupt flag).

My thoughts:

* I find solution (1) OK for the short term but very lame.

* I find solution (2) very interresting.

* I find solution (3) also very interresting. And we do already have set the convention, that the "pointer" of the regbank shall always point to the top [here - follow this link](https://github.com/sy2002/QNICE-FPGA/blob/master/doc/best-practices.md#all-languages): QUOTE:

> When using register banks make sure that the register bank selector in the upper eight bits of SR is always pointing to the highest active bank. Reason: Interrupt Service Routines might interrupt your code any time and they might also use register banks to save and restore the lower registers. So: If you need to reserve multiple banks, then just increase the bank selector accordingly. The principle is similar to reserving something on the stack, but the other way round: You are incrementing the bank selector to reserve space but you would decrement the stack pointer for doing so.

We might need to sharpen this quote a bit to also mention, that you always need to make sure the regbank pointer is "at the top" before calling a sub routine. Would be OK for me, too

bernd-ulmann, 2020-09-14:
To be honest, I would just say that the register banks behave basically like a stack of registers i.e. if a stack entry is released a program can not rely on its validity. You wouldn't work with a stack as you suggested for the register bank, right? ;-) 

You are, of course, right, it is tempting to use more than one register bank and prepare higher entries with values for later use which would get overwritten. But I think, the cleanest way to avoid this is to declare the register bank a register stack which implies that released entries are volatile. :-)

sy2002, 2020-09-14:
> it is tempting to use more than one register bank and prepare higher entries with values for later use which would get overwritten. 

I guess we have a misunderstanding here. It is not about preparing higher entries with values for later use but lower ones:

INCRB
  Fill
INCRB
  Fill more
INCRB
  Fill even more

With a stack, you can absolutely do that as Michael pointed out:

Assume you need 100 bytes on the stack:

SUB 100, SP
MOVE SP, R0

<now start working with the memory you just reserved>

So how could we do that using regbanks?

One idea could be

MRB ("move regbank"):

MRB Regbank, Source Reg in Regbank, DST Reg in currently active bank

The SR always points like the SP to the "top" so all would be good then

But independently from that having a CLI STI would be state of the art, too

bernd-ulmann, 2020-09-14:
I do not think we have a misunderstanding here but obviously different opinions on how to use the register stack. There is no problem in something like

ENTRY  INCRB
  DO SOMETHING
  INCRB
  DO SOMETHING
  DECRB
  DO SOMETHING
  DECRB
  RET

 Even if an interrupt occurs, the ISR will INCRB, do something minding its own business, and then DECRB and so the routine just interrupted can continue at the correct position. 

 What indeed does not work is to "reserve" n entries in the register stack and use them within a routine. If an ISR occurs this will, of course, run havoc.

 But, at least from my perspective, this is not how the register bank should be used. I (personally) would never use it in a non-hierarchical manner, i.e. with properly matched INCRB/DECRB parentheses, so to speak.

 Something different: We only had two spare bits in the SR. If we use one for an instruction trap feature and one for enabling/disabling interrupts, we are left without a bit to distinguish between user/kernel mode!

sy2002, 2020-09-15:
* The topic of CLI/STI is postponed.

* The new feature described in issue #135 might be a good solution for this issue, too.

* Dependiing on when we will tackle issue #135, we might go for option 1) as described in https://github.com/sy2002/QNICE-FPGA/issues/109#issuecomment-691950116. I am citing option 1):

> We could set up the convention, that ISRs **must not** use register bank switching at all but use only the stack.
I guess this might be the easiest / short term "fix". The assembler programmer must not forget this (we need to add it as a best practice in our `doc/best-practices.md`). And: We must also make sure, that C programmers then always add the `__norbank` directive to their `__interrupt` directive - or, even better, besides mentioning this in `doc/best-practices.md` we might want to have a `#define __isr __interrupt __norbank` in our `sysdef.h` and tell everybody to use `__isr` instead of `__interrupt`, so that nobody is ever forgetting this.

* We need to update our Programming Best Practices around ISR programming, but also around how to use register banks

MJoergen, 2020-09-17:
Ok, I made some progress with this issue.

Using the ILA I could see that the hardware stops at address `0x1A16` (`XOR @R2++, R0`) in the `qtransfer` program (the `QCALC_CRC16` subroutine). This instruction attempts to read from address `0xFF12` (the UART Rx FIFO), which causes an infinite wait, because the FIFO is empty at that point.

Some further debugging showed that the register R9 in the loop (`_QCCRC16_LOOP`) had an incorrect value (anything larger than `QBURST_WORDS`) already upon entry to the routine. In fact, just before entry into the loop, an interrupt has just finished processing. The following picture shows the ISR completing (`Int_Active` going low), and then the processor resumes execution at address `0x1091`:

[image]

The relevant output in `monitor.lis` is the following:

```
002758  108C  E0C0        MTH$MULU INCRB
002759  108D  0F80  FF18                  MOVE IO$EAE_OPERAND_0, R0
002760  108F  0802                        MOVE R8, @R0++ ; R0 now points to OPERAND_1
002761  1090  0901                        MOVE R9, @R0
002762  1091  0F80  FF1C                  MOVE IO$EAE_CSR, R0
002763  1093  0F81  0000                  MOVE EAE$MULU, @R0
```

So a little code examination revealed that the `qtransfer` program makes use of the `EAE` module to calculate the buffer size. However, the `fancy` program also uses the `EAE` within its ISR and that won'ẗ work.

The `EAE` is a shared resource (in particular the operands written to it), and the ISR in the `fancy` program modifies these operands, but does not restore them. Therefore, when the ISR returns the `EAE` operands (and therefore the result too) contains incorrect values.

The `fancy` ISR must therefore preserve the state of the `EAE`.

Moreover, there is a similar problem with the use of the `VGA$PALETTE_ADDR`. This register, too, must be preserved if used in an ISR.

I'll update the fancy ISR to preserve the state of the EAE and the `VGA$PALETTE_ADDR`

MJoergen, 2020-09-17:
Oh, and we need to update the doc/best_pratices.md as well.

sy2002, 2020-09-17:
Hi Michael,

WOW - this is a remarkable analysis! Thank you.

Indeed, I was so focused on regbanks and stuff that I completely missed out, that our hardware devices are shared resources and if an ISR is using it, the ISR needs to save all registers of the device beforehand and restore them afterwards.

Good job! Excellent point!

MJoergen, 2020-09-17:
Ok, so far the `fancy.asm` is fixed. I removed the `SYSCALL(mulu)` completely, because that was the easiest in this case.

However, more generally, if an ISR wants to use the monitor `SYSCALL(mulu)` then it is currently not possible, unless the ISR manually saves and restores the EAE registers. On the other hand, having the `SYSCALL(mulu)` perform this save and restore on every call is a waste of precious CPU cycles.

So either we simply state the ISRs are not allowed to use `SYSCALL(mulu)`, or we could perhaps add a "safe" version to the monitor, i.e. a `SYSCALL(mulu_safe)`, which makes sure to save and restore the EAE state. This "safe" function is slower, and is meant to be used only in ISR's (or in other re-entrant code).

What are your thoughts ?

sy2002, 2020-09-17:
Sounds great! Three topics:

### Topic 1: The formula

> I removed the SYSCALL(mulu) completely, because that was the easiest in this case.

OK, I had a look and did a sample calculation: Your new code is not equivalent to `(scanline + j) * 62`, so we either need to adjust the comment or or the code (I would prefer the code, because I feel the older version looked better, but keeping the new code and changing the comment is also OK for me): So what is the formula, that is being calculated now?

```
                MOVE    @R3, R8                 ; (scanline + j) * 62
                ADD     @R0, R8

                MOVE    R8, R10
                SHL     5, R10
                ADD     R8, R10
                SHL     1, R10
```

### Topic 2: ISR-safe operating system funtions

> or we could perhaps add a "safe" version to the monitor, i.e. a SYSCALL(mulu_safe)

I think this is a very good idea. Lets do that.

**I think we need an own section about how to write ISRs in our programming best practices**, where we collect all these things, including your finding in https://github.com/sy2002/QNICE-FPGA/issues/135#issuecomment-693927113

When it comes to naming: I would prefer to use a prefix instead of a postfix, that makes it very clear, that these functions are meant to be used in an ISR, so something like

```
isr_mulu
isr_muls
isr_divu
isr_div
isr_vga_copyfont
... and more ...
```

As it is one of our goals to increase our RAM by reducing the ROM to end before 0x2000 (issue #7) , we need to be cautious to not waste more space than necessary. So I opened a new issue with my proposal: Issue #138

### Topic 3: fancy.asm + qtransfer (?) + cpu_test does not work for me

* I did this test: Started fancy and loaded Q-TRIS via SD-CCard and ran it: worked great.
* Then ended Q-TRIS and used qtransfer to transfer the `cpu_test.asm`
* The CPU test did stop at `84D8`

A reason could be: fancy is still increasing/decreasing the register bank pointer and therefore (even if it is not changing SR itself as you described in https://github.com/sy2002/QNICE-FPGA/issues/135#issuecomment-693927113) fiddling with register banks of the CPU test?

Can you reproduce it?

Might be your suggestion from https://github.com/sy2002/QNICE-FPGA/issues/135#issuecomment-693927113, that from now on, ISR do not need INCRB/DECRB but more like a `MOVE 0x8000, R14` would solve it?

MJoergen, 2020-09-17:
Topic 1 : My bad. It's fixed now.

Topic 3 : The CPU test still uses all 256 register banks. The changes I made were only local, not pushed to repository.

Regarding the CPU test: I consider the existing `cpu_test.asm` to be a **functional** test of the CPU, i.e. testing each feature individually. That includes testing that all 256 register banks are working. But that rules out other kinds of tests, like the one in this issue.

So perhaps we should write a special (new) `stress_test.asm` that tests **combinations** of features, including using a lot of register banks, having multiple interrupts, using various hardware resources (EAE, VGA, etc), plus more.

bernd-ulmann, 2020-09-17:
WOW! That is a really incredibly analysis! 

To be honest, I do not see this as a problem. On all machines I have programmed so far, interrupt routines are pretty tricky and when it comes to external devices which have a state, one must be careful.

When we will (some day :-) ) implement user and kernel mode, all access to external devices, including the EAE, will only be possible in kernel mode (in which interrupts will nevertheless be disabled since an interrupt is the only way to enter kernel mode), so we won't have any problem then. 

As of now the real problem is that user code as well as ISR code can access IO devices.

I do not think that we should think about additional hardware to save states of IO devices. Instead either the ISR should save the state of the EAE or not use it as of now until we have user/kernel mode.

What do you think? :-)

sy2002, 2020-09-17:
@MJoergen I still had the problem, that I cannot load anything from the SD card (like Q-TRIS) nor even browse a directory, while fancy is running. Then it changed both fancy ISRs according to your idea in https://github.com/sy2002/QNICE-FPGA/issues/135#issuecomment-693927113 : **It now works like a charm!** 🥳 

I find your idea of making the best-practice "convention" that something like `MOVE    0xF000, SR` as the new "entry to an ISR" instead of `INCRB` very elegant: This is 15 regbanks for the ISR and 241 for the applications. This still leaves PLENTY of regbanks for the application and everything (including my beloved "reserving" of regbanks) still works like a charm. And I guess 99% of the ISRs will be happy with 15 regbanks.

After having done this I did these stress tests while fancy was running:

* Use the serial terminal to load "adventure.out" via `M` and `L` which is a 163k (!) .out file: ✅ Works like a charm
* Browse the SD Card (directories): ✅ Works like a charm
* Run "adventure.out", "sierpinski.out" and "q-tris.out": ✅ Works like a charm

I guess that this also solves #62 - I just do not have time today to test it, but presumably we just need to update our `test_programs/timer_test.asm`, too.

So I would state:

Given, that the ISR author sticks to our best practices of how to write an ISR, I would not only consider this issue here as closed (and closed it), but I would also have the very warm gut feeling that we are on our way to a perfectly stable and reliable interrupt system now. :-)

closed 2020-09-17 by sy2002