sy2002/QNICE-FPGA #111: assembler: Help a programmer to find an error

sy2002, 2020-09-07:
This issue is a follow-up of https://github.com/sy2002/QNICE-FPGA/issues/106#issuecomment-688203972

When you assemble a larger project that as some include files, an error message like the following example is not helpful, because you have no reference point where to look:

```
assembling: tile_ed.asm
assemble: Line 885: No valid mnemonic/no label (does not start in column 1)
assemble: Line 1279: Unresolved label or equ >>END<<!
main: There were 2 errors during assembly! No files written!
An unrevoverable error occured!
```

A solution might be to write a `.lis` file also in case of error. Or to have an `.err` file or something like that.

@MJoergen and @bernd-ulmann Ideas welcome.

MJoergen, 2020-09-07:
Or perhaps print out the offending string. Then the user can do a simple search in the source file.

sy2002, 2020-09-07:
> Or perhaps print out the offending string. Then the user can do a simple search in the source file.

Also a good idea! Maybe even something that Bernd might want to consider additionally to having some `.err` file?

Because all this info about lines (like `assemble: Line 885: No valid mnemonic/no label (does not start in column 1)`) is currently quite useless to have any file where you can see what `Line 885` actually means ;-)

sy2002, 2020-09-07:
Merging the `develop` branch with `master` now for performing the V1.6 release. That means this issue will go to V1.7.

bernd-ulmann, 2020-09-07:
I changed the assembler so that in the case of an error the complete line of source code is written to stdout. I hope this helps as it is only a quick-fix... :-)

closed 2020-09-07 by bernd-ulmann

sy2002, 2020-09-07:
Thank you, Bernd. I guess this is indeed helpful.