sy2002/QNICE-FPGA #118: Accidentally reading from an empty UART fifo will stall the entire system

MJoergen, 2020-09-08:
The software is required to check the UART status before reading the fifo, so this is not an urgent problem.

sy2002, 2020-09-08:
This whole 2015 UART needs some refactoring, I guess :-)

sy2002, 2020-09-15:
@MJoergen Did your refactoring implicitly solve this?

MJoergen, 2020-10-19:
The question is what behavior do we want? Should the read stall, as it does today? Or should it return 0? Or something else?

Personally, I don't like the idea of the processor stalling while waiting for an outside event. I would like the read to return 0 (or some other value like 0xFFFF) if the FIFO is empty.

sy2002, 2020-10-19:
> Personally, I don't like the idea of the processor stalling while waiting for an outside event.

Me neither, this is why I would consider today's behaviour a hardware bug that I introduced, when I added the FIFO to the UART.

> I would like the read to return 0 (or some other value like 0xFFFF) if the FIFO is empty.

Yes. Returning `0` would be good. I mean: It is a programming error when reading without first knowing if something is in the buffer at all, so returning "trash" (in this case a `0`) is a fair thing to do.

Another question: 

Is it fair to assume that even when fixing this behaviour here, we still need to make sure that the `uart_cpu_ws` signal still is set to high during the time that we are waiting for the FIFO to deliver the value to the register? (This is how the mechaism works today: `reading` triggers `uart_cpu_ws` and `reset_reading` resets `reading` and is asserted as soon as the FIFO signals `fifo_rd_valid`.)

MJoergen, 2020-10-19:
Actually, I rewrote part of the code so that there is no longer a need for `uart_cpu_ws`. This is because the first byte of the FIFO is ready before the actual read is performed. This means the value can be returned to the CPU combinatorially, and then the value is read from the FIFO, thereby preparing the next value for the next read.

closed 2020-10-19 by MJoergen