sy2002/QNICE-FPGA #126: Monitor function `MTH$IN_RANGE_U`: Better way to do it?

sy2002, 2020-09-12:
This function sets carry = 1 if (unsigned) `R8 >= R9 and R8 < R10`:

https://github.com/sy2002/QNICE-FPGA/blob/develop/monitor/math_library.asm#L288

I feel that this is quite a complicated solution. But currently I am not able to see the wood for the trees.

So please check: Is this the fastest and most robust solution? Then close this issue.

Else: Please create the fastest and most robust solution :-)

bernd-ulmann, 2020-09-12:
@sy2002 : Hi Mirko - could you have a quick look at test_programs/math_test.asm if I understood MTH$IN_RANGE_U correctly. As of now (maybe I am a bit too tired) I wonder about the behaviour. The first call sets C with R8 = 0x1000, R9 = 0x1000 and R10 = 0x1000 but shouldn't R9 <= R8 < R10 be satisfied for C to be set?

I changed one comment line in the math-library as well as I at first misinterpreted the intended behaviour of the comparison routine. :-)

sy2002, 2020-09-12:
Hi Bernd, you did not check in test_programs/math_test.asm.

But from your comments I can at least guess, what you tried. Interessting question about the semantics!

You are right that one might want to receive C=0 in such a situation. 

In general this would mean if R9 = R10 then always C=0, right?

Do you want to improve it or do you want me to?

Do you see more room for improvement or that`s it?

sy2002, 2020-09-12:
P.S. EDIT: I forgot to pull. You did check it in.

bernd-ulmann, 2020-09-12:
Huh? I checked in and pushed... ...but in develop! What would have been the correct branch? Sorry!

bernd-ulmann, 2020-09-12:
I will take care of the comparison routine! That is ideal after having written two reports on two lengthy master theses during the last couple of hours. :-) Finally something fun to do! :-)

sy2002, 2020-09-12:
> Huh? I checked in and pushed... ...but in develop! What would have been the correct branch? Sorry!

All good. I forgot to pull :-)

bernd-ulmann, 2020-09-12:
I changed the routine (which only saved one instruction ;-) ) and it now passes the (trivial) test routine in the test_math.asm test program. Could you (@sy2002 ) please have a look if everything is OK?

sy2002, 2020-09-14:
I tested it. Works. You saved one instruction, I like that :-) Thank you for your support. Closed.

closed 2020-09-14 by sy2002