sy2002/QNICE-FPGA #128: Feature Request: Select the UART baud rate on the fly

MJoergen, 2020-09-14:
Currently, the baud rate is hardcoded at the value 115 kbit/s. The Nexys4DDR hardware is easily capable of running at any baud rate up to approx 3 Mbit/s. Furthermore, the FTDI driver in Linux is also capable of selecting any baud rate in the same range.

However, most terminal programs only allow setting a baud rate from a very limited set of values. Therefore, we should not hard-code a specific fast baudrate.

Instead, I would like the baudrate to be programmable, either via a software register write, or via one of the hardware switches on the Nexys4DDR board.

A minimum acceptable solution could be a single bit to select between a fast (1 Mbit/s) and a slow (115 kbit/s) mode.

One thing to consider is that the baud rates achieveable  in the current hardware design (based on dividing the CPU clock of 50 MHz by 16 and then by an integer divider) gives possible baud rates that only very badly overlaps with the list of baud rates supported by e.g. the program `picocom`. So for instance selecting a divisor of 3 gives the baud rate 50/16/3 = 1.04 Mbit/s. This therefore is a 4 % sampling error compared to the value 1 Mbit/s supported by `picocom`. This particular value seems to work without problems (regardless of Issue #125). Other divisors lead to baud rates that don't fit as well with the possible values supported by `picocom`.

bernd-ulmann, 2020-09-14:
Classic UARTs use quartz oscillators with rather strange frequencies which match better typical baud rates. Just out of curiosity: Would be PLL an option? One might use the 50 MHz system clock as the base frequency and then derive exactly matching clock rates that perfectly match typical baud rates, what do you think? Are PLLs possible on FPGAS? (Probably a stupid question but my PLLs were always setup with discrete parts. :-) )

sy2002, 2020-09-14:
This is a really good idea! I would opt for a combination of hardware switches and software registers: You need the hardware register to get started up the system but then you can use a bit in some software register to change the baud rate and to deactivate/override the hardware switch.

About the clock speeds that can be barely created using a simple integer clock divider: Just as Bernd said:

What about MMCM / PLL: The Artix-7 is having multiple of them onboard. We could use one of them to create just the right clock speed and then use your clock-domain-crossing magic to get it handled.

Of course we need to make sure, that we are still a portable solution, but then, or example, in systems that don't have (that much or that powerful) MMCMs, we just offer the 115 kbit/s and on other systems we offer more.

Our to be defined SYSINFO register (issue #101) can provide info, what is supported by the hardware.

MJoergen, 2020-09-15:
So, I refactored the UART code and implemented a programmable divisor, with a 16 times better resolution than before, and without the need for an MMCM/PLL.

I've tested by writing 0x0032 to 0xFF10 (via the terminal program picocom), then dynamically change the baudrate to 1000000, and continue typing just like before. Going back is just as easy: Write 0x01B2 to 0xFF10, and then change the baudrate to 115200, and we're back where we started.

I have (somewhat arbitrarily) limited the possible baudrates to the range 12 k - 3.1 M.

I've verified that transferring (and running) the `cpu_test.out` file works in both cases.

BTW, the speed change is very much noticeable!

Pressing the CPU reset button on the Nexys4DDR board resets the baudrate to 115200.

@sy2002 : How and where should we document this?

sy2002, 2020-09-15:
This is a great achievement. I love performance improvements a lot! :-) Here is my suggestion of how to document this:

* The main documentation should go to `doc/IO_Devices.md`. (BTW: You might want to update VGA, as it is not accurate any more)

* doc/README.md: I added three `TODO @MJoergen` stubs into this file, please document here, too

closed 2020-09-17 by MJoergen