sy2002/QNICE-FPGA #130: Documentation: Enjoy and celebrate, then double-check, then close this issue :-)

sy2002, 2020-09-14:
Gentlemen, we can allow ourselves a little bit if celebration: In the meantime we do have a quite nice (pun intended :-)) amount of documentation. This is why I restructured doc/README.md and added something that resembles closely to a searchable index at the end of a book. 

Enter **Full Documentation in Alphabetical Order** 😄 

Please fire up your webbrowser and look here:

https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/README.md#full-documentation-in-alphabetical-order

Obviously I deliberately have some redundancy in this list, so that when you search for something you are more likely to find it.

@bernd-ulmann and @MJoergen : Please check if I forgot something important. After that, please close this issue.

MJoergen, 2020-09-14:
Nice job!

Suggestion: I would like to move the sections "Specifics of the Nexys 4 DDR and Nexys A7 hardware" and "Specifics of the MEGA65 hardware" to the hw/README.md, since they refer to specific hardware platforms.

sy2002, 2020-09-14:
Good idea. Yes, please do. Be aware, that there are a lot of "cross-links" within various `.md` files, so you might want to `grep` that after you did the change and adjust it.

bernd-ulmann, 2020-09-14:
WOW! The new documentation is awesome! :-)

MJoergen, 2020-09-15:
Done!

closed 2020-09-15 by sy2002